ROBOTIS-GIT/DynamixelSDK #109: read problem with no status packet

JingYao0512, 2017-08-19:
- Which DYNAMIXEL is it?

  [　XL-320　] 

- Which CONTROLLER is it?

  [　OpenCM　] 

- Which VERSION of DynamixelSDK is it?

  [　3.4.7　] 

I used raspberry 3 to send packet via OpenCM9.04  to XL-320, and I already tested to successfully
write the packet to XL-320 (for example, setting LED color, setting motor position). However, reading the information from motor can't be worked. Error message was like "There is no status packet! ".

I can't find the problem. Have anyone encountered the same problem?

Thanks,
Jing-Yao

ROBOTIS-Leon, 2017-08-20:
@JingYao0512

Hi,

so you made its own firmware of signal transformer right?

I think that motor responds with both write and read instruction packets, and must be returning the status packet, but was not properly written in the buffer of your pc, since OpenCM perhaps returned very slowly and your PC couldnt wait until full status packet comes into. 

1. Did you use RPI3 GPIO baudrate at 115200? and OpenCM either? 

2. Find LATENCY_TIMER in the port handler, then increase the value

give me feedback after doing it

JingYao0512, 2017-08-21:
@ROBOTIS-Leon 
Hi,

First, Thanks your advise.
     1. RPI3 GPIO didn't be used for my project. and baudrate of baudrate is 1M. 
     2. I tried to set LATENCY_TIMER to 30, but it encountered the same problem. 

Second, I try to using  RoboPlus to update my controller. However, it didn't work.
So I try to scan my dxl, I used the "dxl_monitor" to test, the following link is result.
http://imgur.com/dAhdzk1

Third, the result of read function as following link. It's really weird. The only reason which I can think is there is no great connection between OpenCM, XL-3210 and raspi.
http://imgur.com/7gtFV6Q
This is my code, I revise the example of broadcast_ping.c then adding the read and write function. 
http://imgur.com/M4LFvFX

※The "DEVICENAME" is "/dev/ttyACM0", and I used the PROTOCOL_VERSION2 because of XL-320.

ROBOTIS-Leon, 2017-08-21:
@JingYao0512 

I think the firmware you set in the OpenCM should be modified or done something.

Does the OpenCM change write mode <-> read mode well?

Try checking if the status packet comes to read buffer on your PC or not, modifying some code in the port handler. printf all data in the buffer came from OpenCM

JingYao0512, 2017-08-21:
@ROBOTIS-Leon 

Yes , I write "Tosser" to OpenCM9.04. "Tosser" is a function of ROTOBIS IDE. It can send packet from coding to XL-320. 

I connect OpenCM and  PC by Mini-USB, and "Tosser" is written to OpenCM. I already can send and write the packet to dynamixel by dynamixelSDK on Windows. Now, I want to use the same concept on raspi, but it fail.

I will try your advise, then giving you feedback. 
Thank you

ROBOTIS-Leon, 2017-08-21:
@JingYao0512 

Hmm.. no idea currently, since ROBOTIS IDE is kinda not fully tested one and will be replaced with Arduino IDE official DynamixelSDK package.
but it seems theres no way but to test it.
Let me do it, a week? maybe?

JingYao0512, 2017-08-21:
OK, I will test and try it, too.
If I have any solution, I will post it.

Thank you!

dudu2683, 2017-08-27:
Hi

I tried to follow the instruction on YouTube- "Episode 4.2: ReadWrite C++ Example - Windows"
 im runing the read_write.cpp on VS2015 but I'm getting a " There is no status packet "

I'm using the MX-106 

How I can solve it?

ROBOTIS-Leon, 2017-08-28:
@dudu2683 

Hi,

next time, write the issue on `New Issue`.
I couldn't find anything related with current issue.

BTW, did you try modifying some in the codes?
Protocol 1.0 readwrite example uses write2ByteTxRx() and read2ByteTxRx() functions because Protocol 1.0 MX series position control needs 2 byte numbers, but Protocol 2.0 MX series position control is available by using 4 byte numbers, so you should use write4ByteTxRx() and read4ByteTxRx() instead, I guess.

JingYao0512, 2017-08-31:
@ROBOTIS-Leon
Hi, Leon
I already solve this problem. I changed its USB setting to full-speed. I tried a lot of setting but failed for last weeks like changing baud rate, changing OS on raspi3, update firmware of opencm & dynamixels.

Sending packet to dynamixel always is succeed, so I think it maybe UART port problem. I follow this method from this website(https://ludovicrousseau.blogspot.tw/2014/04/usb-issues-with-raspberry-pi.html). Now, I can read packet from dynamixel well.

I change its command to my setting.
`sudo nano /boot/cmdline.txt`
then 
`dwc_otg.lpm_enable=0 console=ttyACM0,115200 kgdboc=ttyACM0,115200 dwc_otg.speed=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`

Thanks

ROBOTIS-Leon, 2017-08-31:
@JingYao0512 

you did very good work man

Now I remember what I did in my old work.
I guess I also set those stuffs before using RPI 2 UART communication XD

I will put this in FAQ list

謝謝

closed 2017-09-05 by ROBOTIS-Leon