ROBOTIS-GIT/DynamixelSDK #111: mx-12w

cod65, 2017-09-03:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?
MX 12w
  [　-　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?
usb2dynamixel
  [　　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?
3.4.7
  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　　　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

cod65, 2017-09-05:
ok, well I realized my port wasnt set properly.

changed to:

DEVICENAME                  = "COM3".encode('utf-8')# Check which port is being used on your controller
                                                            # ex) Windows: "COM1"   Linux: "/dev/ttyUSB0"

now I get:

>>> ================================ RESTART ================================
>>> 
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected
Press any key to continue! (or press ESC to quit!)
Traceback (most recent call last):
  File "C:\Users\SPACE1970\Documents\DynamixelSDK-master\python\protocol1_0\read_writeHACK01.py", line 139, in <module>
    if getch() == chr(ESC_ASCII_VALUE):
  File "C:\Users\SPACE1970\Documents\DynamixelSDK-master\python\protocol1_0\read_writeHACK01.py", line 51, in getch
    return msvcrt.getch().decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
>>>

cod65, 2017-09-05:
commented out line 139:
    # if getch() == chr(ESC_ASCII_VALUE):
     #   break

and it started running!

but why?

ROBOTIS-Leon, 2017-09-05:
@cod65 

Hi,

good work.

I have no idea since I've never seen this error before and doesn't become realized in my testing PC
but I've tracked some issue that must be related with your issue,
so let's start trying some other ways like replace ESC value to another.

cod65, 2017-09-07:
I replaced to:

ESC_ASCII_VALUE             = 0x31

which presumably is the '1' key, if I understand correctly.

No difference, as  python cursor returns to '>>> '  before i enter a key, regardless.

ROBOTIS-Leon, 2017-09-08:
@cod65 

Hi,

it seems like it doesn't match up with the value of the key, essentially.
Since I can't see the same error as you saw, and is perhaps not able to be shown in my computer though you send me your code, this time you should solve it (I will help you as much as I can). 

Dig more about python system, and let me know how you fixed it after all.

Sorry for that and thank you for your patience

closed 2017-11-30 by ROBOTIS-Leon