ROBOTIS-GIT/DynamixelSDK #112: Dynamixel RX24F core dumped??

Angit16, 2017-09-03:
Hi, 

I initially used the read_write.cpp code in the example provided in the SDK on RX-24F. The servo was working. Since last week I noticed the following problem.
```
ahar88@MU00061481L:~/DynamixelSDK/c++/example/protocol1.0/read_write/linux64$ ./read_write
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected
Press any key to continue! (or press ESC to quit!)
[TxRxResult] Protocol does not support This function!
Segmentation fault (core dumped)
ahar88@MU00061481L:~/DynamixelSDK/c++/example/protocol1.0/read_write/linux64$
```
This has happened when I used Dynamixel-sdk 3.4.7. I tried to debug it, but could not get yet. Can you please help me out?

Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　RX-24F　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　　PC via USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　3.4.7　　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2017-09-05:
@Angit16 

Hi, again

can you try version 3.5.1 first?

and show me some codes, if it is possible

Angit16, 2017-09-05:
Hi, 

I forgot to make from build initially. So, libdxl_x64_cpp.so was not generated. It is solved now and works with version 3.5.1.

ROBOTIS-Leon, 2017-09-06:
@Angit16 

Great,

please let me know if you have another question

closed 2017-09-06 by ROBOTIS-Leon