ROBOTIS-GIT/DynamixelSDK #113: How to set joint or wheel mode for RX-24F?

Angit16, 2017-09-04:
Hi,
I am using Dynamixel RX-24F. I have following queries.
1. How to set the joint or wheel mode for this servo? I did not find any specific instruction for the same.
2. I use ubuntu 14.04 LTS. What should be the latency timer value in port_handler_linux.h?

Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　-RX-24F　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　PC via USB2Dynamixel　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　　3.4.7　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2017-09-05:
@Angit16 

Hi, again

1. Go <kbd>http://support.robotis.com/en/</kbd>→<kbd>Product Information</kbd>→<kbd>Actuator</kbd>→<kbd>Dynamixel</kbd>→<kbd>RX Series</kbd>→<kbd>RX-24F</kbd> then see CCW Angle Limit in the Control table. 

It describes how to change the drive mode of the Dynamixel

2. I think that 4 msec was enough in Ubuntu 14.04.

Angit16, 2017-09-05:
Hi,
I checked the manual before also. My question is if I set the minimum and maximum angle limit between 10 and 1023, should it work in joint mode? Because normally these two limits were set at 100 and 4000 respectively in 'read_write.cpp'. They were providing present positions as well. 

Also, I am not very clear about whether I can control velocity with RX-24F. It will be of great help if you could provide some information on this.

thanks,

ROBOTIS-Leon, 2017-09-05:
@Angit16 

the value set in the code is just a range of where you want to turn
the value 100 and 4000 is just for setting goal position, not the angle limit
if you want to turn it from 400 to 800, set cw and ccw limit via robo plus or putting extra code to set position control mode before it writes actual goal position

Angit16, 2017-09-06:
Hi,
I set the Min and Max position as 400 and 700 respectively. According to my understanding the following piece of code should rotate the servo back and forth between 400 and 700 starting from any arbitrary position in the beginning. But, it does following
```

while(1)
    {
        printf("Press any key to continue! (or press ESC to quit!)\n");
        if (getch() == ESC_ASCII_VALUE)
            break;

        // Write goal position
        dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_MX_GOAL_POSITION, dxl_goal_position[index], &dxl_error);
        if (dxl_comm_result != COMM_SUCCESS)
        {
            printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
        }
        else if (dxl_error != 0)
        {
            printf("%s\n", packetHandler->getRxPacketError(dxl_error));
        }

        do
        {

            // Read present position
            dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, ADDR_MX_PRESENT_POSITION, &dxl_present_position, &dxl_error);
            if (dxl_comm_result != COMM_SUCCESS)
            {
                printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
            }
            else if (dxl_error != 0)
            {
                printf("%s\n", packetHandler->getRxPacketError(dxl_error));
            }
            printf("[ID:%03d] GoalPos:%03d  PresPos:%03d\n", DXL_ID, dxl_goal_position[index], dxl_present_position);
           
        }while((abs(dxl_goal_position[index] - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD));

        // Change goal position
        if (index == 0)
        {
            index = 1;
        }
        else
        {
            index = 0;
        }
    }
```
Results on terminal:
```
result:	400	699
result:	400	697
result:	400	694
result:	400	688
result:	400	682
result:	400	675
result:	400	668
result:	400	659
result:	400	651
result:	400	641
result:	400	630
result:	400	621
result:	400	610
result:	400	601
result:	400	591
result:	400	579
result:	400	572
result:	400	562
result:	400	552
result:	400	541
result:	400	531
result:	400	521
result:	400	511
result:	400	502
result:	400	494
result:	400	483
result:	400	475
result:	400	466
result:	400	456
result:	400	446
result:	400	436
result:	400	424
result:	400	414
result:	400	405
Press any key to continue! (or press ESC to [quit!)](url)
result:	700	400
*** Error in `./read_write': malloc(): memory corruption: 0x000000000086e070 ***
Aborted (core dumped)
ahar88@MU00061481L:~/DynamixelSDK-3.5.1/c++/example/protocol1.0/read_write/linux64$ ./read_write
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press ESC to quit!)
result:	400	699

```
Next time when I start running it the present position shows 699, where as last time it was stopped at 400.

I am trying to understand if this code is supposed to rotate the servo back and forth or not.
 Your help is highly appreciated.

regards,

ROBOTIS-Leon, 2017-09-06:
@Angit16 

Yes, you're right.

Can you show me the code? via google drive or something if you have difficulty

ROBOTIS-Leon, 2017-09-06:
@Angit16 

As I check your code,

the error occurs in File Writing but not from the example code.
Check the code I sent

Angit16, 2017-09-06:
My objective: Run the servo with constant speed (changeable) back and forth between two given position values.

Another important question: How can I access the servo time, so that I could plot position vs. time?

ROBOTIS-Leon, 2017-09-06:
@Angit16 

You can handle speed in position control mode by changing MOVING SPEED (ADDR : 32 / 33).
Set those values by write2ByteTxRx().

RX-24 can be controlled single time per 8msec, in common, so if you want to split position control by many times, calculate `split_number = [how much long you want to rotate it] / 0.008` then calculate (goal_pos - present_pos) / split_number. write2ByteTxRx this, then sleep(8) for next write

Check the reference available [here](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_friends/turtlebot3_realturtlebot/turtlebot3_realturtlebot.ino)

ROBOTIS-Leon, 2017-09-06:
@Angit16 

You can change packet return time delay value in the [port_handler_linux.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_friends/turtlebot3_realturtlebot/turtlebot3_realturtlebot.ino)

Angit16, 2017-09-10:
Hi,
I could change the speed periodically during the servo operation as you suggested.
But, I could not change the return delay time. It prints 250 always. By the way, I modified LATENCY_TIMER =4 as I am using Ubuntu 14.04.

One more thing:

I noticed that a small section in "port_handler_linux.cpp" code which is given below provides overflow in time information.

> double PortHandlerLinux::getCurrentTime()
> {
> 	struct timespec tv;
> 	clock_gettime( CLOCK_REALTIME, &tv);
> 	return ((double)tv.tv_sec*1000.0 + (double)tv.tv_nsec*0.001*0.001);
> }

Just to check how it works, I ran a small piece of code

> #include <stdio.h>
> #include <fcntl.h>
> #include <string.h>
> #include <unistd.h>
> #include <termios.h>
> #include <time.h>
> #include <sys/time.h>
> #include <sys/ioctl.h>
> #include <linux/serial.h>
> #include <iostream>
> 
> using namespace std;
> 
> int get_TIME()
> {
>     struct timespec tv;
>     clock_gettime( CLOCK_REALTIME, &tv);
>     return ((double)(tv.tv_sec*1000.0) + (double)(tv.tv_nsec*0.001*0.001));
> }
> 
> int main()
> {
> double time_s;
> time_s=get_TIME();
> cout << "time now is:" << time_s << endl;
> return 0;
> }
> 

and it produces the output 
```

Starting /home/ahar88/opengle_test/build/build-test4-Desktop_Qt_5_7_0_GCC_64bit-Default/test4...
time now is:-2.14748e+09
```

Then I modified the code portion as below.

> int get_TIME()
> {
>     struct timespec tv;
>     clock_gettime( CLOCK_REALTIME, &tv);
>     return ((long)(tv.tv_sec*1000.0) + (long)(tv.tv_nsec*0.001*0.001));
> }
> 
> 
> int main()
> {
> long time_s;
> // double time_ns;
> time_s=get_TIME();
> cout << "time now is:" << time_s << endl;
> return 0;
> }

and it produces output as 
```
Starting /home/ahar88/opengle_test/build/build-test4-Desktop_Qt_5_7_0_GCC_64bit-Default/test4 exited with code 0

Starting /home/ahar88/opengle_test/build/build-test4-Desktop_Qt_5_7_0_GCC_64bit-Default/test4...
time now is:1765407222
```

Can you please have a look into it?

ROBOTIS-Leon, 2017-09-11:
@Angit16 

Isn't it since return time delay of your Dynamixel is set as 250.
and I will test as you described after Sep, 30th, since some work for exhibition, IROS, will happen.

I hope you can wait with patience.
Thank you

ROBOTIS-Leon, 2017-11-30:
@Angit16 

Hi,
I'm looking at what you pointed out

actually, changing double into long doesn't make any problem.
but as it calculates tx_time_per_byte by dividing baudrate, it seems like it needs to be as double.

Do you have any idea or problem

closed 2017-12-11 by ROBOTIS-Leon