ROBOTIS-GIT/DynamixelSDK #115: write negative values for position offset and comunication speed

chuchodel, 2017-09-08:
- Which DYNAMIXEL is it?
MX-28AT

- Which CONTROLLER is it?
Win 10, Matlab via USB2Dynamixel

- Which VERSION of DynamixelSDK is it?
SDK 3.5.1,  protocol 2.0
_____________________________________________________________________________

I need to set a home offset on each Dynamixel. I use the next instruction:

write4ByteTxRx(Port,2,servo_ID,ADDRESS_HOME_OFFSET,value_offset);

But when value_offset<0  (negative values), the instruction crashes.

In addition, I would like to reduce the communication latency with the Dynamixels. Now I am using a BuadRate=3Mbs, I set the COM delay to the minimum in windows (1ms) and RoboManager 2,0 (time delay=0). With that setting, I get a speed of 10-6 ms. Could I reduce this value?

Thanks
Jesus

ROBOTIS-Leon, 2017-09-11:
@chuchodel 

Hi,

Thank you and I will test it.
but since some work for exhibition, IROS, and national holidays will happen,
the works will be carried out after Oct. 20th

I hope you can wait with patience.
Thank you

ROBOTIS-Leon, 2017-11-30:
@chuchodel 

Hi,
and sorry for the late. I hope that still you are working on this

did you change the latency timer value in port_handler_windows?

ROBOTIS-Leon, 2017-12-11:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-12-11 by ROBOTIS-Leon

itsameWolf, 2025-03-20:
I have the same Issue every time I send a negative value to home offset using the write4ByteTxRX function it will convert the value from signed to unsigned.