ROBOTIS-GIT/DynamixelSDK #117: No status pack error with DynamixelPro

tr-gis, 2017-09-13:
Hello,
I have two dynamixel pro motors (H54-100-S500-R and H54-200-S500-R)attached in  a pan and tilt configuration. I modified the read-write.cpp file (DynamixelSDK/c++/example/protocol2.0/read_write.cpp
) to set DXL_ID to 2 and  BAUDRATE = 1000000. I also changed DXL_MINIMUM_POSITION_VALUE =     0 and DXL_MAXIMUM_POSITION_VALUE to -250950 to match my configuration. 

I created a variable dxl_current_goal_position=DXL_MINIMUM_POSITION_VALUE and am decrementing by 1000 (so 0,-1000,-2000 ...) and writing it as goal position.Then I ran the make command inside DynamixelSDK/c++/example/protocol2.0/read_write/linux32 and ran the executable read_write which gives the following output : 

 ```
/home/user/DynamixelSDK/c++/example/protocol2.0/read_write/linux32
./read_write
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] There is no status packet!
[ID:003] GoalPos:000  PresPos:000
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] There is no status packet!
[ID:003] GoalPos:-1000  PresPos:000

```

However the motor fails to respond and the output keeps printing the same message 

> [TxRxResult] There is no status packet!
> [ID:003] GoalPos:-1000  PresPos:000 

until I kill it using Ctrl+C. I changed the baud rate from 57600 to 1M after going through this [post](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/64l). The motors respond when I give them goal position from R+ manager in Windows 8. Could you please help me resolve this issue ?
 
Here is a [link](https://github.com/tr-gis/C-plus-plus-sample-code/blob/master/read_write.cpp) to the modified read_write.cpp file
PS: I am using Ubuntu 16.04 on an intel i3 system.

ROBOTIS-Leon, 2017-09-14:
@tr-gis 

Hi,

firstly check if it runs when you put positive value into the goal position.

but since some work for exhibition, IROS, and national holidays will happen,
helping you will be carried out after Oct. 20th

I hope you can wait with patience.
Thank you

tr-gis, 2017-09-14:
Hi @ROBOTIS-Leon,

Thank you for the response. Like you suggested I did try with positive values but I end up getting the same output. Did you mean Sep. 20th ?.I can see that IROS is scheduled around then.

ROBOTIS-Leon, 2017-09-18:
@tr-gis 

Hi, and sorry for my late reply

Yes, and theres holiday week after IROS, so it is hard to work for the issues until October, 20th.

but if any time I can take a breath, I will deal with the issues

sorry and thank you for understanding

hthomas60, 2017-09-27:
I am having a simmilar issue. I have two motors, an MX-28 and AX-12. The MX-28 works when I run the example code read_write.cpp however the AX-12 gives the "No status packet error". I belive the AX-12 motor is working though because it responds to comands from the Dynimexle Wizard.

ROBOTIS-Will, 2017-09-29:
@hthomas60 
Hello, please make sure that both MX-28 and AX-12 are using an identical Protocol Version.
If MX-28 firmware version is higher than v39, it will use Protocol 2.0 whereas AX-12 only supports Protocol 1.0.
If you want to use both Dynamixels at the same time with Protocol 1.0, please set the Protocol Version(13)(http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-28(2.0).htm#bookmark6) of MX-28 to 1.
Thank you.

hthomas60, 2017-10-01:
Thank you for your reply. I am using Protocol 1.0 for both motors. When I try to run /DynamixelSDK/c++/example/protocol1.0/read_write/linux64$ ./read_write with the AX-12 plugged in I get

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)

If I unplug the AX-12 and plug in the MX-28 and run the same program it works fine.

tr-gis, 2017-10-03:
@ROBOTIS-Leon The original issue I opened is solved. It seems the problem was with the servo. Replacing it has resolved the issue.I guess can close this issue if @hthomas60's issue is fixed too. 

-Thanks

ROBOTIS-Leon, 2017-11-24:
@tr-gis 

Hi, 
and sorry for the late. I hope that your work was done great

@hthomas60 

Hi, 
and sorry for the late. I hope that still you are working on this

Since your issue is being dealt in #126, this issue will be closed

closed 2017-11-24 by ROBOTIS-Leon

ROBOTIS-Will, 2018-03-29:
@JoyChopra1298, @ludusrusso 
If you are running AX series, please note that your default baudrate is 1Mbps.
C or C++ examples?
Protocol 1.0 or 2.0?
What are the Dynamixel settings? (baudrate, ID, Status return level)
What port is assigned to USB2Dynamixel or U2D2? (COM?? or ttyUSB??)

ycean, 2018-08-07:
Hi, 
I face the same error. Here is my test details:
Dynamixel model: MX 106
Connected with pc using USB2Dynamixel
Running with Ubuntu 16.04

The error message is:
**/catkin_ws/src/DynamixelSDK/c/example/protocol1.0/read_write/linux64./$ ./read_write
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)**

I am trying to setup my servo according to the manual : **http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/device_setup/**

I did change the define detail according to the control table for MX106 in my read_write.c
The coding below is my amendment for my read_write.c:

**#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>
#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table
#define ADDR_MX_TORQUE_ENABLE           34                  // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION           30
#define ADDR_MX_PRESENT_POSITION        36

// Protocol version
#define PROTOCOL_VERSION                1.0                 // See which protocol version is used in the Dynamixel

// Default setting

#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        57600
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0                 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4095                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     10                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b**

is there any solution it?

Thanks.

ycean, 2018-08-08:
Sorry that I have no idea why I "unassigned" it, I was editing my question at that moment.

robotpilot, 2018-08-08:
Hi @ycean, 
You don't have to care. However, this issue is a closed issue. Please open a new issue.
I think the error name is the same but it is a different situation.

ycean, 2018-08-08:
Thanks with noted.

prefpkg21, 2019-01-24:
Check your value of the Return Delay Time. The config yaml files try to set it to 0. I found increasing to 50 or above minimizes packet error. I'm running MX64-T to MX28-T via OpenCR board.

EduardoLuisSantos, 2019-04-08:
> Hi,
> I face the same error. Here is my test details:
> Dynamixel model: MX 106
> Connected with pc using USB2Dynamixel
> Running with Ubuntu 16.04
> 
> The error message is:
> **/catkin_ws/src/DynamixelSDK/c/example/protocol1.0/read_write/linux64./$ ./read_write Succeeded to open the port! Succeeded to change the baudrate! [TxRxResult] There is no status packet! Press any key to continue! (or press ESC to quit!)**
> 
> I am trying to setup my servo according to the manual : **http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/device_setup/**
> 
> I did change the define detail according to the control table for MX106 in my read_write.c
> The coding below is my amendment for my read_write.c:
> 
> **#if defined(**linux**) || defined(**APPLE**) #include <fcntl.h> #include <termios.h> #define STDIN_FILENO 0 #elif defined(_WIN32) || defined(_WIN64) #include <conio.h> #endif**
> 
> **#include <stdlib.h> #include <stdio.h> #include "dynamixel_sdk.h" // Uses Dynamixel SDK library**
> 
> **// Control table #define ADDR_MX_TORQUE_ENABLE 34 // Control table address is different in Dynamixel model #define ADDR_MX_GOAL_POSITION 30 #define ADDR_MX_PRESENT_POSITION 36**
> 
> **// Protocol version #define PROTOCOL_VERSION 1.0 // See which protocol version is used in the Dynamixel**
> 
> **// Default setting
> 
> #define DXL_ID 1 // Dynamixel ID: 1
> #define BAUDRATE 57600
> #define DEVICENAME "/dev/ttyUSB0" // Check which port is being used on your controller
> // ex) Windows: "COM1" Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"**
> 
> **#define TORQUE_ENABLE 1 // Value for enabling the torque
> #define TORQUE_DISABLE 0 // Value for disabling the torque
> #define DXL_MINIMUM_POSITION_VALUE 0 // Dynamixel will rotate between this value
> #define DXL_MAXIMUM_POSITION_VALUE 4095 // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
> #define DXL_MOVING_STATUS_THRESHOLD 10 // Dynamixel moving status threshold
> 
> #define ESC_ASCII_VALUE 0x1b**
> 
> is there any solution for it?
> 
> Thanks.

Note the previous comments, for AX-12 you should change your boud rate to 1000000