ROBOTIS-GIT/DynamixelSDK #118: Control Motor Speed

LoFurtado, 2017-09-14:
Hi! I'm using the read.write program in Matlab with the MX-106, but I could not control the speed. Can you help me with this control?

(Sorry for any typing errors. I'm Brazilian)

ROBOTIS-Leon, 2017-09-18:
@LoFurtado 

Hi, 

1. Did you read the [Dynamixel SDK Wiki](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki) which shows how to use it?

2. Sure, that's our job, but theres no information that can be based on our suggestion. Can you attach the details that you have? Codes you are testing / Error log picture or something you've taken, etc.

3. because of some schedules here, it is hard to follow up whole issues here until October, 20th.

but if any time I can take a breath, I will deal with the issues

sorry and thank you for understanding

GrigoriyAY, 2017-10-18:
For example, you can use command:  _**calllib('dynamixel', 'dxl_write_word',ID,32,value)**_ for library Dynamixel SDK 3.x.

ROBOTIS-Leon, 2017-11-24:
@LoFurtado 

Hi,
I'm sorry for the late

1. Is your MX106 set in Protocol 2.0 or still in Protocol 1.0? Did you update the firmware of your MX106?
You can check this by RoboPlus

2. If you are able to use read_write example in MATLAB, put following codes in the example source code routine you aimed

- protocol 1.0
``` matlab
    write2ByteTxRx(port_num, 1.0, DXL_ID, 32, typecast(int16([the value you want to set]), 'uint16'));
```

- protocol 2.0
``` matlab
    write4ByteTxRx(port_num, 2.0, DXL_ID, 104, typecast(int32([the value you want to set]), 'uint32'));
```

Try this and ask me further
Thanks

closed 2017-12-01 by ROBOTIS-Leon