ROBOTIS-GIT/DynamixelSDK #120: Bug in protocol1_packet_handler.cpp, line 222

pramilo, 2017-09-21:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　ALL　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　　3.5.2　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [ I attach
[protocol1_packet_handler.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/1320781/protocol1_packet_handler.zip)
[protocol1_packet_handler.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/1320783/protocol1_packet_handler.zip)

 a commented file ](　Link the SOURCE SAMPLES here, if necessary　)

Summary:

In line 222 of protocol1_packet_handler.cpp, you check if the error mask returned is valid.
However your check has a bug. Your code is:

rxpacket[PKT_ERROR] >= 0x64

**The problem is that 0x64 = ‭01100100‬,  which 

does not cover all valid masks**. (see http://support.robotis.com/en/product/actuator/dynamixel/communication/dxl_packet.htm )

I think you meant to say 64 (without the 0x) but that _would also be wrong_, because it would not account for situations with multiple errors at once.

The correct check should be for bitmask 01111111‬ = 127 or 0x7F in Hexadecimal.

The line should become:
 rxpacket[PKT_ERROR] > 127 (NOTE: just ">" not ">=")

This is **causing the SDK to give error of "No reply packet" even if the servo replies but has an error condition** of Voltage Error, Angle Limit Error, Checksum or Overload Error. IN all cases, the SSK will say the servo did not reply because of this Wrong check.

pramilo, 2017-09-21:
I must add that this check (validity of the error mask) is somewhat redundant. Other bus devices using the Dynamixel protocol could make use of the full byte.
Furthermore discarding a packet that is correct in length and checksum just because it fails the error mask verification seems a little aggressive.

Failing the checksum of LEN would seem much more severe and worth discarding the packet.

ROBOTIS-Leon, 2017-11-30:
@pramilo 

Hi, how are you
long time no see

You are right.

Thanks for sharing.

closed 2017-11-30 by ROBOTIS-Leon