ROBOTIS-GIT/DynamixelSDK #121: Failed to set 1000000 baudrate in Mac OSX Matlab

dmrobotix, 2017-09-22:
I get an error whenever I try to set the baudrate to 1000000. All other baud rates work on my setup, not sure what is going on. I am able to set the baudrate using my lab mate's computer which runs Windows.

Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　DXL-320 ]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　　MacOSX / Matlab via USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　3.5.1　DynamixelSDK-3.5.1　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [no](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2017-10-20:
@dmrobotix 

Hi,
sorry for the late reply

I also was suffering from the limit: Mac OS X doesn't support the communication with the baudrate over 230400, as everyone mentions.

So, every library, every example source code of the Dynamixel for Mac OS X is only available if you use supported baudrate.

If you have any ideas, please let me know

ROBOTIS-Leon, 2017-11-24:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-11-24 by ROBOTIS-Leon

shardulamrite, 2020-06-09:
@dmrobotix Hey. 
I too am using a MacOS and trying to control the Dynamixel XM 430 with MATLAB via OpenCM 9.04 C. 
Could you please tell me how you managed to build the C library with the 'makefile' in the repository ?
Thanks