ROBOTIS-GIT/DynamixelSDK #122: generating full packet with CRC in Python

cod65, 2017-09-26:
Spent weeks trying various ways to control position of protocol 1 and 2  motors over serial.

I understand that with 'tosser' sketch or USB2Dynamixel :

(0xFF,0xFF,0xFD,0x00,0x01,0x09,0x00,0x03,0x74,0x00,0x00,0x00,0x00,0x00,0xCA,0x89)

sends a protocol 2 motor to position 512 in registry.  This works.
I understand   last 2 hex bytes are generated CRC 16 code.
I assume that this is the best way to get as close to realtime performance as possible, by sending HEX to register, plus the CRC code would insure the most likely uncorrupted packet.

I'm looking for a way, via Python to create complete packets in int or  HEX.
I cant use the majority of example code  becasue my application is not multithreaded, apparently.

Essentially at this point in time  I am able to send the above  example 'pos 512' HEX code, with CRC, but I dont understand how to generate my own CRC encoded commands .
I found a Python module PyCRC to do this, in int or HEX, but I don't understand what part of my command packet to run through this module.

any help would be greatly aprec.

## example goto pos 512 ##(0xFF,0xFF,0xFD,0x00,0x01,0x09,0x00,0x03,0x74,0x00,0x00,0x00,0x00,0x00,0xCA,0x89)

from PyCRC.CRC16 import CRC16
## example input format:      input = b'\x01\)x09\x03'
print(CRC16().calculate(input))

Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　XM430 W210R　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　　USB2Dyna, OpenCM9.04　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　　　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Will, 2017-09-29:
Hello,
First of all, sending Goal Position 512 to Dynamixel ID 1 with protocol 2.0 will be as below;
0xFF, 0xFF, 0xFD, 0x00, 0x01, 0x09, 0x00, 0x03, 0x74, 0x00, 0x00, **0x02**, 0x00, 0x00, 0xCA, 0x89
The last two bytes are CRC-16 based on CRC-16 (IBM/ANSI)(There are many CRC calculation algorithms; http://crccalc.com/).
The CRC calculation source code can be found at e-Manual
http://support.robotis.com/en/product/actuator/dynamixel_pro/communication/crc.htm

cod65, 2017-09-29:
thks Will. I had found the CRC source code, but I am a Python novice, at best, and I assumed that source is C or C++ ? ( or agnostic ?). I'm not much of  a programmer, so was hoping to use a Python module  that would do CRC16IBMANSI. 
I dont really know what to do with that source in order to adapt it to Python (if it even needs porting).
IDLE isnt recognizing unsigned short for starters, so I assume this is C and I would need to adapt to Python(which is beyond me). Ive been diggin in the Python bsection of the SDK to try and fish out just a CRC 16  encoder that I can use in an environment that has an embedded version of Python as scripting language. thks for your attention.
regards.

cod65, 2017-09-29:
and thks for taking the time to find what is a possible typo in my '512' example - that could be a big part of the issue, obviously ;)

ashsifat, 2017-11-20:
Hi, did you figure out the code for CRC in python from the SDK. I am trying to do the same thing now. Please let me know if you have. Thanks

cod65, 2017-11-20:
Hi, no sorry I took a different path and used the existing Python code, without digging too much deeper into it.In this manner I use the CRC coder that is already written, without creating the packet from scratch

ashsifat, 2017-11-20:
Did you use the sdk python code? Can you please elaborate which existing python code gives you the solution for protocol 2.0? I couldn’t figure out how to use the sdk code for protocol 2. I was trying to use it with the dynamixel controller package that you can find on github. I am close to finishing my method but don’t know if it will work. Please let me know your method. Can you by any chance share your package? Thanks a lot.

cod65, 2017-11-21:
I'm away from my lab until Dec 2. Until then i can only recall from memory. Essentially im using the Python examples in SDK, which create the packet and CRC check, so you dont have to re-create that stuff. It worked for prot 1 and 2 motors. Check in with me then and I will take a look if youdtill need help

cod65, 2017-11-21:
I'm away from my lab until Dec 2. Until then i can only recall from memory. Essentially im using the Python examples in SDK, which create the packet and CRC check, so you dont have to re-create that stuff. It worked for prot 1 and 2 motors. Check in with me then and I will take a look if you still need help

ashsifat, 2018-01-09:
Hi, could you please let me know which functions in the sdk you use and how you call the SDK functions in your own package? I solved the CRC problem, but it seems to be a lot simpler if I could use the sdk functions in python to actuaute the motors. Tried to call the ping function by including the sdk library in my working package. but that didn't work. Thank you very much.

cod65, 2018-01-12:
Hi, I 've had to shelve this project for a bit, so I'm a bit out of the loop. essentially I found that I was able to control 'goal position'  by hacking in a variable sent over either UDP, or TCP/IP. I run the python as an 'applet' in the background receiving a variable over either of these protocols. (there were pros and cons to either- I cant remember what they were). But my control app(Touch designer) seems to work on Python in a 'frame based' manner , rather than fully multi  threading.... not sure I know what I'm talking about, but that seems to be the case. So, essentially any variable you want to change use existing SDK code and something like UDP or TCP input in place of variable I use 'socket' and 'struct' python modules to do this. . Sorry, that is all I can give you for now - I'm wrapped up in other projects ... this one is kind of taken down for a bit while I tend to other things. If I get back into it I will revisit this thread. Dog speed !

cod65, 2018-01-12:
ps the main advantage was I didn't have to worry about packet forming, because the existing SDK code takes care of all of that in my applet.

ROBOTIS-Leon, 2018-01-17:
@cod65 

Hi, how is your work going

I guess I will start the work for Dynamixel SDK Python, which is a 100% python code with the low level SDK API.

I just wanted to know how far you did already.

Cheers

ashsifat, 2018-01-17:
Hello,
Thank you for letting me know. I am now trying to make a controller in
protocol 2. I am not a good software engineer, rather a EE, so I am trying
to directly manipulate the packets structures using the dynamixel
controller packages codes. To be specific, I have modified the dynamixel_io
python code to work for protocol 2 and have so far made the sync write
commands working. Now I am trying to make a joint state controller.
But it would be great if I could understand what you guys are doing.

Thanks

Regards,
Ashrarul Haq Sifat
Graduate Student
Bradley Dept. of ECE
Virginia Tech

ROBOTIS-Leon, 2018-04-09:
@ashsifat

Hi, I couldn't know if you had commented here long time ago. Sorry for that.

Now we started making DynamixelSDK purely for Python. It will have the APIs almost as same as another languages have. I will report the progress when anything is completed.

Cheers.

ashsifat, 2018-04-09:
@ROBOTIS-Leon 
That's great news. can you please let me know when you finish? I will need to use it on my new system. It would make things a lot easier. Thank you so much.

cod65, 2018-04-09:
Yes, i think this python focus will reinvigorate my interest in Dynamixels as well! Thks for the work, Leon.

cod65, 2018-04-09:
Im personally primarily interested in  reliable remote 'realtime' dynamixel control from my pc- based python scripting environment

ROBOTIS-Leon, 2018-04-17:
@ashsifat

the work will be finished in this month. I should call everyone who will be helpful on this project for the beta test. So I will call you here and get your advice.

@cod65

Hi, again
I'm glad to hear it. The topic which has been asked everytime is the 'realtime'; I want to know how it would be calculated and verified, so I think that we should discuss further during the betatest.

Thank you guys, and I'm expecting that DynamixelSDK Python will be developed as follow the SDK C++ convention, but I'm wondering if it is good idea or not. I am also suffering from not knowing how to upload it into PyPi pip. I will be glad if I can get some related details from python experts like you guys.

cod65, 2018-04-17:
Hi Leon,
I'm not even close to an 'expert' in Python or programming, but look forward to being involved in any beta testing through Python

ashsifat, 2018-04-17:
Hi,

I will be testing my new system with the new dynamixel actuators soon. It
will be a perfect opportunity for testing. But I also am not an expert
specially about the PyPi pip thing you are talking about. So, some
direction about the usage of it would help a lot. Thank you.

Regards,
Ashrarul Haq Sifat
Graduate Student
Bradley Dept. of ECE
Virginia Tech

ROBOTIS-Leon, 2018-04-26:
@ashsifat @cod65 

Hi, 

I just uploaded the pythonized DynamixelSDK in develop branch DynamixelSDK/python. Please have a look, and give me advices. Currently I'm trying to extend its compatibility with Win/Mac and Python 3.

roncapat, 2018-04-26:
I found a bit of repeated errors along the code. In the next hours I'll submit a PR with port_handler_linux.py and protocol1_packet_handler.py fixes as an example for fixing the other files, with a brief description of what is wrong.

ROBOTIS-Leon, 2018-04-26:
@roncapat 

I'm appreciated of that

roncapat, 2018-04-26:
#182

roncapat, 2018-04-26:
By the way, I suggest to stick to PEP8 recommendations regarding style. Some IDEs like Pycharm can help enforcing those styles rules and detect subtle bugs effectively. In the Python world, PEP8 is a very important thing.

ROBOTIS-Leon, 2018-04-27:
@roncapat 

Great idea, 
I also once wondered whether python also has its standards on its code, but never known PEP8

okay, ROS also intoduces its code style as follow the PEP8 as well,
great information. I should refactor codes

ROBOTIS-Leon, 2018-04-27:
@roncapat 

can you reply some checks in PR if you don't mean?

ROBOTIS-Leon, 2018-05-04:
@roncapat 

Can you check if the sync/bulk read write works well?

roncapat, 2018-05-04:
I'll check

roncapat, 2018-05-04:
Python style recommendations says that function and method name should be lowercase (not camelcase). I don't refactored them, but I'd suggest to think about this change.

roncapat, 2018-05-04:
After reviewing the code, now I'll start to test it. I'll keep you informed.

roncapat, 2018-05-04:
I built a simple script to check if all was working. 
I started with PacketHandler. Protocol 1 had some bugs:
* reg_write was crashing, because of an offset-by-one error
* the subsequent instruction sent after action() was failing, because txRxPacket wasn't collecting a response for action packets. 

Fixes pushed in the PR above

roncapat, 2018-05-04:
I would propose also to alias these methods, or renaming them:
 * `syncWriteTxOnly` -> `syncWriteTx`
 * `bulkWriteTxOnly` -> `bulkWriteTx`

Why specifying `Only` if there is already a distinction between `tx`, `rx` and `txrx` (I think that this is the distinction that you wanted to point to)? The actual naming schema is quite confusing, at a first glance it seems that packet_handler has `only` a piece and misses something, although it's not.

roncapat, 2018-05-11:
@ROBOTIS-Leon, any news? Did you see my PR?

ROBOTIS-Leon, 2018-05-14:
@roncapat 

I'm sorry for my late.
I was completely knocked out from turnover.

Let's talk in your PR... I guess here's no focused on the activity anymore.

@cod65  @ashsifat 

Hey, guys
we will go on to the PR #185 
So, see you there.

closed 2018-05-14 by ROBOTIS-Leon