ROBOTIS-GIT/DynamixelSDK #123: [TxRxResult] Incorrect status packet!

gugudela, 2017-09-28:
- Which DYNAMIXEL is it?

  [　AX-12A　]  

- Which CONTROLLER is it?

  [　CM700　] 

- Which VERSION of DynamixelSDK is it?

  [　Sorry,I don't know. But it's the latest for linux　] 

 I did exactly as the video showed. 
However, when I ran the example, it told me that [TxRxResult] Incorrect status packet!  And then it entered an infinite loop.. like this:

[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
 [TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
.......
 
How to solve this problem, please help me. Waiting for your reply, thank you.

ROBOTIS-Will, 2017-09-29:
@gugudela 
Hello,
If you have seen this video(https://youtu.be/pHC5p3d6NM4), the example is intended to be use with USB2Dynamixel and PC.
If you are using CM-700 for connecting your AX-12A to PC, the example might not work properly.
Thank you.

ROBOTIS-Leon, 2017-11-24:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

Babtou67, 2017-11-28:
Hello, 

I'm using an USB2Dynamixel with AX-12A Dynamixel motor, alimentation 12V, i follow this 3 videos:

https://www.youtube.com/watch?v=RLnpps8eh6U&list=PLEf1s0tzVSnSgVzf4AREpat_P_HRLSiDn&index=2
https://www.youtube.com/watch?v=J0xIwmxg-MQ&index=8&list=PLEf1s0tzVSnSgVzf4AREpat_P_HRLSiDn
https://www.youtube.com/watch?v=zb8I05D-LGE&list=PLEf1s0tzVSnSgVzf4AREpat_P_HRLSiDn&index=5

but i got this errors: 

LIFEBOOK-NH532:~/DynamixelSDK-3.5.3/c++/example/protocol1.0/read_write/linux64$ ./read_write 
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] Incorrect status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] Incorrect status packet!
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] Incorrect status packet!
[....]

On Windows i'm able to see the motor with wizard manager and to make them turn using the goal control. 

What can i do to make them work on Linux ?

Thanks

ROBOTIS-Leon, 2017-11-28:
@Babtou67 

Hi,
are you using the example source code as it was downloaded? or did you modify something in the code?

and make sure that the ID and baudrate of the motor are properly set in the example source code

Babtou67, 2017-11-28:
Yes, i juste changed the baudrate from 57600 to 1000000 for the AX-12A

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller

Otherwise, i didn't touch anything.

I verify the ID of the motor with Wizard manager it is 1.

ROBOTIS-Leon, 2017-11-28:
@Babtou67 

and did you push the slide button to ttl communication in the USB2Dynamixel?

Babtou67, 2017-11-28:
Yes it is on TTL.

I also try with ROS, following the controller_manager tutorial i get this kind of errors:

[INFO] [1511880116.037192]: pan_tilt_port: Pinging motor IDs 1 through 3...
[ERROR] [1511880116.231207]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511880116.247398]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511880116.263203]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511880116.279106]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511880116.295293]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[INFO] [1511880116.300242]: pan_tilt_port: Found 0 motors , initialization complete.

ROBOTIS-Leon, 2017-11-28:
@Babtou67 

Okay, I will check it tomorrow morning again,
but I actually tested it yesterday XD

Babtou67, 2017-11-28:
It works now, i put the baudrate with Wizard manager to 57600 bu there is still some errors:

~/DynamixelSDK-3.5.3/c++/example/protocol1.0/read_write/linux64$ ./read_write 
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:100  PresPos:509
[ID:001] GoalPos:100  PresPos:499
[ID:001] GoalPos:100  PresPos:485
[ID:001] GoalPos:100  PresPos:469
[ID:001] GoalPos:100  PresPos:451
[ID:001] GoalPos:100  PresPos:434
[ID:001] GoalPos:100  PresPos:413
[ID:001] GoalPos:100  PresPos:394
[ID:001] GoalPos:100  PresPos:371
[ID:001] GoalPos:100  PresPos:350
[ID:001] GoalPos:100  PresPos:329
[ID:001] GoalPos:100  PresPos:308
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:308
[ID:001] GoalPos:100  PresPos:261
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:261
[ID:001] GoalPos:100  PresPos:216
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:216
[ID:001] GoalPos:100  PresPos:171
[ID:001] GoalPos:100  PresPos:156
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:156
[ID:001] GoalPos:100  PresPos:109
Press any key to continue! (or press ESC to quit!)
[RxPacketError] Out of range error!
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100
[ID:001] GoalPos:4000  PresPos:100

still impossible to launch it with ROS:

process[dynamixel_manager-1]: started with pid [2974]
[INFO] [1511881637.669545]: pan_tilt_port: Pinging motor IDs 1 through 3...
[ERROR] [1511881637.864063]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511881637.880158]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511881637.896216]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511881637.912125]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[ERROR] [1511881637.928206]: Exception thrown while getting attributes for motor 1 - Invalid response received from motor 1. Wrong packet prefix ['\x00', '\xff']
[INFO] [1511881637.933038]: pan_tilt_port: Found 0 motors , initialization complete.

with a baudrate of 57600

ROBOTIS-Leon, 2017-11-28:
@Babtou67 

Great job!
Change the value of MAXIMUM POSITION from 4000 to lower than 1023 in the first place,,, since the maximum position value of the AX-12 is 1023. The value 4000 is intended for the use of MX series.

and, try to change latency timer in port_handler_linux.cpp to more than 16 ms

Babtou67, 2017-11-28:
I put 1023 in read_write.cpp and 32 ms in port_handler_linux.cpp but i still get this:

~/DynamixelSDK-3.5.3/c++/example/protocol1.0/read_write/linux64$ ./read_write 
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:100  PresPos:1021
[ID:001] GoalPos:100  PresPos:1020
[ID:001] GoalPos:100  PresPos:1009
[ID:001] GoalPos:100  PresPos:995
[ID:001] GoalPos:100  PresPos:979
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:979
[ID:001] GoalPos:100  PresPos:938
[ID:001] GoalPos:100  PresPos:921
[ID:001] GoalPos:100  PresPos:901
[ID:001] GoalPos:100  PresPos:880
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:880
[ID:001] GoalPos:100  PresPos:833
[ID:001] GoalPos:100  PresPos:814
[ID:001] GoalPos:100  PresPos:793
[ID:001] GoalPos:100  PresPos:770
[ID:001] GoalPos:100  PresPos:749
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:749
[ID:001] GoalPos:100  PresPos:698
[ID:001] GoalPos:100  PresPos:681
[ID:001] GoalPos:100  PresPos:659
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:659
[ID:001] GoalPos:100  PresPos:610
[ID:001] GoalPos:100  PresPos:593
[ID:001] GoalPos:100  PresPos:569
[ID:001] GoalPos:100  PresPos:548
[ID:001] GoalPos:100  PresPos:525
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:525
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:525
[ID:001] GoalPos:100  PresPos:446
[ID:001] GoalPos:100  PresPos:434
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:434
[ID:001] GoalPos:100  PresPos:383
[ID:001] GoalPos:100  PresPos:365
[ID:001] GoalPos:100  PresPos:343
[ID:001] GoalPos:100  PresPos:322
[ID:001] GoalPos:100  PresPos:299
[ID:001] GoalPos:100  PresPos:279
[ID:001] GoalPos:100  PresPos:255
[ID:001] GoalPos:100  PresPos:232
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:100  PresPos:232
[ID:001] GoalPos:100  PresPos:184
[ID:001] GoalPos:100  PresPos:167
[ID:001] GoalPos:100  PresPos:144
[ID:001] GoalPos:100  PresPos:123
[ID:001] GoalPos:100  PresPos:103
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:1023  PresPos:101
[ID:001] GoalPos:1023  PresPos:107
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:1023  PresPos:107
[ID:001] GoalPos:1023  PresPos:138
[ID:001] GoalPos:1023  PresPos:151
[ID:001] GoalPos:1023  PresPos:168
[ID:001] GoalPos:1023  PresPos:188
[ID:001] GoalPos:1023  PresPos:209
[ID:001] GoalPos:1023  PresPos:227
[ID:001] GoalPos:1023  PresPos:249
[ID:001] GoalPos:1023  PresPos:270
[ID:001] GoalPos:1023  PresPos:290
[ID:001] GoalPos:1023  PresPos:311
[ID:001] GoalPos:1023  PresPos:332
[ID:001] GoalPos:1023  PresPos:354
[ID:001] GoalPos:1023  PresPos:374
[ID:001] GoalPos:1023  PresPos:399
[ID:001] GoalPos:1023  PresPos:418
[ID:001] GoalPos:1023  PresPos:441
[ID:001] GoalPos:1023  PresPos:461
[ID:001] GoalPos:1023  PresPos:484
[ID:001] GoalPos:1023  PresPos:508
[ID:001] GoalPos:1023  PresPos:530
[ID:001] GoalPos:1023  PresPos:553
[ID:001] GoalPos:1023  PresPos:573
[ID:001] GoalPos:1023  PresPos:596
[ID:001] GoalPos:1023  PresPos:618
[ID:001] GoalPos:1023  PresPos:639
[ID:001] GoalPos:1023  PresPos:661
[ID:001] GoalPos:1023  PresPos:682
[ID:001] GoalPos:1023  PresPos:706
[ID:001] GoalPos:1023  PresPos:726
[ID:001] GoalPos:1023  PresPos:750
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:1023  PresPos:750
[ID:001] GoalPos:1023  PresPos:799
[ID:001] GoalPos:1023  PresPos:816
[ID:001] GoalPos:1023  PresPos:837
[TxRxResult] Incorrect status packet!

ROBOTIS-Leon, 2017-11-28:
@Babtou67 

1. Retry to update the firmware in AX-12
2. Get the latency timer back (I think that this can't be thought as a problem anymore)
3. Change the baudrate in the both side (Dynamixel configuration wise and Example code wise)

Babtou67, 2017-11-29:
1. I put latency back to 8ms
2. Baud rate to 1000000 for Dynamixel and example code 

Everything working fine no errors anymore.

But still impossible to connect with ROS.... no idea why...

ROBOTIS-Leon, 2017-11-30:
@Babtou67 

You mean dynamixel-workbench doesn't work?

Babtou67, 2017-11-30:
@ROBOTIS-Leon 

I follow this tutorial: http://wiki.ros.org/dynamixel_controllers/Tutorials/ConnectingToDynamixelBus

One motor AX-12A, the USB2Dynamixel, ID 1, Baudrate: 1000000, ttyUSB0

I sudo chmod 666 /dev/ttyUSB0 so should be fine

But i get this error: 

PARAMETERS
 * /dynamixel_manager/namespace: dynamixel_control...
 * /dynamixel_manager/serial_ports/dxl_tty1/baud_rate: 1000000
 * /dynamixel_manager/serial_ports/dxl_tty1/max_motor_id: 30
 * /dynamixel_manager/serial_ports/dxl_tty1/min_motor_id: 1
 * /dynamixel_manager/serial_ports/dxl_tty1/port_name: /dev/ttyUSB0
 * /dynamixel_manager/serial_ports/dxl_tty1/update_rate: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    dynamixel_manager (dynamixel_controllers/controller_manager.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[dynamixel_manager-1]: started with pid [4374]
[INFO] [1512032262.275141]: dxl_tty1: Pinging motor IDs 1 through 30...
[FATAL] [1512032264.804645]: dxl_tty1: No motors found.
================================================================================REQUIRED process [dynamixel_manager-1] has died!
process has died [pid 4374, exit code 1, cmd /home/terabee/catkin_ws/src/dynamixel_motor/dynamixel_controllers/nodes/controller_manager.py __name:=dynamixel_manager __log:=/home/terabee/.ros/log/33c5ed58-d5ac-11e7-a652-9c4e368b67d8/dynamixel_manager-1.log].
log file: /home/terabee/.ros/log/33c5ed58-d5ac-11e7-a652-9c4e368b67d8/dynamixel_manager-1*.log
Initiating shutdown!
================================================================================
Any ideas ? 

Thanks for your help !

ROBOTIS-Leon, 2017-11-30:
@Babtou67 

Hi, 
you can use it actually,
but I can help you only when you use [dynamixel-workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench/), which is also a official example package for ROS and it uses DynamixelSDK. What you are using is neither using SDK nor ROBOTIS official ROS package.

closed 2017-12-11 by ROBOTIS-Leon

alexisutc, 2019-11-27:
Hello,

i'm facing the same problem :
`[TxRxResult] Incorrect status packet!`

I'm using the dynamixel workbench in ros. So basically i start the node and i got this :

`[ INFO] [1574858977.352412572]: Name : deux, ID : 2, Model Number : 12
[ INFO] [1574858977.356064870]: Name : quatre, ID : 4, Model Number : 12
[ INFO] [1574858977.359673210]: Name : trois, ID : 3, Model Number : 12
[ INFO] [1574858977.362450403]: Name : un, ID : 1, Model Number : 12
[ INFO] [1574858977.500291124]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1574858977.500466176]: [DynamixelDriver] Succeeded to add sync write handler
[ERROR] [1574858978.004020952]: [TxRxResult] Incorrect status packet!
[ERROR] [1574858978.072226858]: [TxRxResult] There is no status packet!
[ERROR] [1574858978.145851471]: [TxRxResult] Incorrect status packet!
[ERROR] [1574858981.353259468]: [TxRxResult] Incorrect status packet!
[ERROR] [1574858982.220028446]: [TxRxResult] Incorrect status packet!`

config :
Raspberry pi (and Navio2) working with raspbian
also I dont have 
`cat: /sys/bus/usb/devices/usb1/latency_timer: No such file or directory`

I'm using the USB2AX since it is for a drone and also is the smallest one

alexisutc, 2019-11-27:
Also sometimes it crashes with 
`groupSyncWrite addparam failed`

Musab-Imadeldin, 2022-12-05:
Hi, have you found a solution to this @alexisutc ? I'd really appreciate if you send how you solved this