ROBOTIS-GIT/DynamixelSDK #125: Error in loadlibrary

marcohui95, 2017-09-29:
I am using dynamixel XH430-W210, however, when I run the matlab code, it always shows:

Error using loadlibrary
No supported compiler or SDK was found. You can install the freely available MinGW-w64 C/C++ compiler;
see Install MinGW-w64 Compiler. For more options, visit
http://www.mathworks.com/support/compilers/R2017a/.

Error in loadlibrary

Error in read_write (line 62)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h',
    'addheader', 'packet_handler.h');
 
I have already followed the software preparation on Youtube, setting up the solution of the output in the file and set paths. Do anyone have ideas how to fix it?

aeer92, 2017-10-11:
I have the same issue with my dynamixel MX-12W, same error message appears and I followed all the steps in the tutorials and videos. What does the error mean and how do we solve it?

GrigoriyAY, 2017-10-18:
Hi, I had simular problem. Then I solved it by installing Microsoft Visual Studio Professional 2013. On some reason I couldn't use Microsoft Visual Studio 2015. May be it can help you.

ROBOTIS-Leon, 2017-10-20:
@GrigoriyAY 

Thank you for your feedback

@marcohui95 @aeer92 

Hi,
sorry for the late reply

As @GrigoriyAY mentioned, some (but most) cases were solved by using DynamixelSDK library file that is  built in the other version of compilers.
Try some other <kbd>Platform toolset</kbd> option described [here](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/3.2.2.1-CPP-Windows)

In the Matlab, there were so many issues in version compatibility with Microsoft Visual Studio everytime. 

I should find the exact reason of them anyway.

closed 2017-11-30 by ROBOTIS-Leon

ZHUIA, 2018-01-13:
did u manage to fix it @marcohui95 ..i am facing a similar problem

ROBOTIS-Leon, 2018-01-15:
@ZHUIA 

Hi,

@GrigoriyAY said that he solved the problem by building new library file

I hope it helps you

ZHUIA, 2018-01-15:
i tried using other [platform toolset] options like VS 2015 but still same problem..  error is shown in pic below
[image]
any other suggestion please