ROBOTIS-GIT/DynamixelSDK #126: AX-12A No Status Packet Linux

hthomas60, 2017-10-01:
Hello,
I am having problems with a AX-12A motor.
I am running c++ and linux64 and am using Prototcol1.0 to communicate. 
The motors are controlled by my PC through a custom ttl converter. (Have used this converter with MX-28s for years with no problems and can control the AX-12 though the Dynamixel Wizard just fine)
I am using Version 3 of the DynamixelSDK.
I only have one motor plugged in.

When I try to run the example code ./read_write on an AX-12A I get the following message.

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!

I get the same response if I try the other example programs such as ping.
If I unplug the AX-12A and plug in an MX-28 the examples work.
I can control the AX-12 and receive data when I run the motor from the Dynamixel Wizard in windows.

Any help would be greatly appreciated.
Thanks!

ROBOTIS-Leon, 2017-10-20:
@hthomas60 

Hi,
sorry for the late reply

At first, please note that we can't guarantee the functions of the Dynamixel with any devices that is developed externally, though its reliability is verified in somehow, someway.

I think that it should run, if you didn't modify any of the codes. 
One thing which should be checked is the maximum & minimum goal position: AX and MX has different limitation of the position. 
Or another thing is communication baudrate, packet return delay time.

I need more clues to find what problem is.
Feel free to comment about what you've found out

hthomas60, 2017-10-20:
Hi Leon,

I have purchased a new AX-12 and moved over to windows. Now my AX-12 appears to work fine however I run the example the MX-28 receives the message.  

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!

What type of clues would you be looking for?

ROBOTIS-Leon, 2017-10-20:
@hthomas60 

Hi, again

simplest way to find the problem is to let me know what kind of situation your test was being done: i.e. source code you have used, video of the problem causion.

I want to test in the same circumstance where you are, and focus on some variables.

hthomas60, 2017-10-20:
I have included the source code as well as an image of my setup and error `message.
My setup
`https://drive.google.com/open?id=0B7lB7luc2WesTUg2cFM1OUN3RDA`

My Error
`https://drive.google.com/open?id=0B7lB7luc2Wesb2oybEdWUW1CRFk`

Source Code
`/*******************************************************************************
* Copyright (c) 2016, ROBOTIS CO., LTD.
* All rights reserved.
*
* Redistribution and use in source and binary forms, with or without
* modification, are permitted provided that the following conditions are met:
*
* * Redistributions of source code must retain the above copyright notice, this
*   list of conditions and the following disclaimer.
*
* * Redistributions in binary form must reproduce the above copyright notice,
*   this list of conditions and the following disclaimer in the documentation
*   and/or other materials provided with the distribution.
*
* * Neither the name of ROBOTIS nor the names of its
*   contributors may be used to endorse or promote products derived from
*   this software without specific prior written permission.
*
* THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
* AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
* IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
* DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
* FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
* DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
* SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
* CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
* OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
* OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
*******************************************************************************/

/* Author: Ryu Woon Jung (Leon) */

//
// *********     Read and Write Example      *********
//
//
// Available DXL model on this example : All models using Protocol 1.0
// This example is tested with a DXL MX-28, and an USB2DYNAMIXEL
// Be sure that DXL MX properties are already set as %% ID : 1 / Baudnum : 34 (Baudrate : 57600)
//

#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>

#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_MX_TORQUE_ENABLE           24                  // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION           30
#define ADDR_MX_PRESENT_POSITION        36

// Protocol version
#define PROTOCOL_VERSION                1.0                 // See which protocol version is used in the Dynamixel

[... 159 lines cut ...]
      index = 1;
    }
    else
    {
      index = 0;
    }
  }

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }

  // Close port
  portHandler->closePort();

  return 0;
}
`

hthomas60, 2017-10-24:
Update: The problem apears to only be in Linux. I can run the AX-12 motors fine using windows but do not get a TX signal in Linux. Unfortunately I need to use Linux for this application but the information might help you find the error.

ROBOTIS-Leon, 2017-11-24:
@hthomas60 

hi, 

I just run your code, and my AX-12 runs perfectly. In the picture of the link you attached shows like you are using the other kind of signal converter, aren't you?

and plus, did you checked your Dynamixel's baudrate?

In my case, my AX-12 baudrate is set to 57600 bps, so I changed the code 
`#define BAUDRATE 1000000` to `#define BAUDRATE 57600`

ROBOTIS-Leon, 2017-12-01:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

MathiasThor, 2018-02-13:
Hallo @hthomas60 and @ROBOTIS-Leon.
I know that the last response to this issue was several months ago, but thought that I would present my solution to the problem anyways. The issue is that the read_write.cpp program is not written for every dynamixel servo, meaning that you have to change both the control table addresses and max/min position. 

As an example I had to change the following addresses:
`#define ADDR_PRO_TORQUE_ENABLE          64
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132`

which originally was:
`#define ADDR_PRO_TORQUE_ENABLE          562
#define ADDR_PRO_GOAL_POSITION          596
#define ADDR_PRO_PRESENT_POSITION       611`

Hope it helps!
- Mathias Thor

AminMohammadi, 2018-02-15:
@MathiasThor 
@ROBOTIS-Leon 
Hello
I'm facing same problem. I follow tutorial videos on wiki, but I get this message:
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
I don't know what is the problem. I checked default settings and protocol version, and they match my motor settings. I have a XM430-W210-T, and I use Dynamixel-SDK-3.5.4. I use USB2Dynamixel.
@MathiasThor   How did you choose those numbers for control table section?
Any help is appreciated.

ROBOTIS-Leon, 2018-02-19:
@MathiasThor 

Thank you for your contribution.
What you've shown will be a great help!

ROBOTIS-Leon, 2018-02-19:
@AminMohammadi 

I'll reopen this issue to easily get solution of yours

AminMohammadi, 2018-02-21:
@ROBOTIS-Leon 
I got my answer. Thanks

ROBOTIS-Leon, 2018-02-21:
@AminMohammadi 

Great

closed 2018-02-21 by ROBOTIS-Leon

cckieu, 2019-06-20:
I got the same error because my dynamixel wasn't properly powered ( the AC adapter plug was unplugged lol)

GameTL, 2025-01-17:
I found this to work on my macbook

I may need to change the address later to align to the [emanual](https://emanual.robotis.com/docs/en/dxl/ax/ax-12a/) but atleast it moves 
I added this 

```
elif MY_DXL == 'AX-12':
    # ADDR_TORQUE_ENABLE          = 64
    # ADDR_GOAL_POSITION          = 116
    # ADDR_PRESENT_POSITION       = 132
    ADDR_TORQUE_ENABLE          = 24
    ADDR_GOAL_POSITION          = 30
    ADDR_PRESENT_POSITION       = 36
    DXL_MINIMUM_POSITION_VALUE  = 100   # Refer to the Minimum Position Limit of product eManual
    DXL_MAXIMUM_POSITION_VALUE  = 1000    # Refer to the Maximum Position Limit of product eManual
    BAUDRATE                    = 1000000
```
changed protocol from 2.0 to 1.0
```
PROTOCOL_VERSION            = 1.0
```