ROBOTIS-GIT/DynamixelSDK #127: Implementing velocity control

tr-gis, 2017-10-03:
I am using Dynamixel SDK 3.5.1 version on Ubuntu 16.04 32 bit version and trying to implement velocity control on MX-64AT. I modified read_write.cpp example file under protocol1.0. The idea I got from this [thread](https://answers.ros.org/question/211471/dynamixel-wheel-mode/) is to set CW and CCW to 0 which enables wheel mode.  Then writing goal speed to the control table should make the servo rotate for given velocity (am I right ?). 

```
#define ADDR_DXL_CW_ANGLE_LIMIT		6
#define ADDR_DXL_CCW_ANGLE_LIMIT	8
#define ADDR_DXL_DRIVE_MODE		10
#define ADDR_MX_TORQUE_ENABLE           24          
#define ADDR_MX_GOAL_POSITION           30
#define ADDR_DXL_GOAL_SPEED		32
#define ADDR_MX_PRESENT_POSITION        36
#define ADDR_DXL_PRESENT_SPEED		38

dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_DXL_CW_ANGLE_LIMIT, 0, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    printf("Dynamixel CW angle set to 0 successfully \n");
  }

dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_DXL_CCW_ANGLE_LIMIT, 0, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    printf("Dynamixel CCW angle set to 0 successfully \n");
  }

while(1)
  {
    printf("Press any key to continue! (or press ESC to quit!)\n");
    if (getch() == ESC_ASCII_VALUE)
      break;

    // Write goal position
    dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_DXL_GOAL_SPEED, 0.6, &dxl_error);
    if (dxl_comm_result != COMM_SUCCESS)
    {
      printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
    }
    else if (dxl_error != 0)
    {
      printf("%s\n", packetHandler->getRxPacketError(dxl_error));
    }

do
    {
      // Read present velocity
      dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, ADDR_DXL_PRESENT_SPEED, &dxl_present_velocity, &dxl_error);
      if (dxl_comm_result != COMM_SUCCESS)
      {
        printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
      }
      else if (dxl_error != 0)
      {
        printf("%s\n", packetHandler->getRxPacketError(dxl_error));
      }

      printf("[ID:%03d] Goal:%f  Pres:%f\n", DXL_ID, 0.6, dxl_present_velocity);

    }while((abs(1 - dxl_present_velocity) > DXL_MOVING_STATUS_THRESHOLD));
}

```

However the motor is not responding. The following is the output when I run the executable : 

```
dynamixel_sdk/example/protocol1.0/read_write/linux32$ ./read_write 
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Dynamixel CW angle set to 0 successfully 
Dynamixel CCW angle set to 0 successfully 
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:0.500000  PresPos:-1.458958
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:0.500000  PresPos:-1.458958
Press any key to continue! (or press ESC to quit!)

```

What could be the cause ?. Also how to determine the range of velocity for any given dynamixel ?

tr-gis, 2017-10-04:
Ok . I got it to work . If anyone else is also looking at the same, refer to the control table documentation for you motor. The MX-64AT's can be found [here.](http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-64at_ar.htm#Actuator_Address_20) Scrolling down to moving speed-> wheel mode section has the necessary details. I think I was giving too small values, the range is 0~2047.

Kindly close this issue.

ROBOTIS-Leon, 2017-10-20:
@tr-gis 

Hi,
sorry for the late reply

Yes, you have followed to the right direction. 
Thank you for your kind feedbacks.

closed 2017-10-20 by ROBOTIS-Leon

ahmaddidiks, 2020-10-23:
how to do this in protocol 2 with python?