ROBOTIS-GIT/DynamixelSDK #129: How to set the dynamixel to wheel mode?

gugudela, 2017-10-18:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　AX-12A　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　PC via USB2Dynamixel　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

Like the title said, I want to set the motor to wheel model, but I don't know how. When I set the motor to wheel mode and tested it with the read_write example(cpp), it didn't work. 
Do you have an example of the wheel mode?
Thank you.

ROBOTIS-Leon, 2017-10-20:
@gugudela 

Hi,
sorry for the late reply

Yeah, since there are no example codes for changing control modes available, you should check its functions and make your own source code. Please check <kbd> support.robotis.com </kbd> → <kbd> Product Information </kbd> → <kbd> Actuator </kbd> → <kbd> Dynamixel </kbd> → <kbd> AX Series </kbd> → <kbd> AX12/AX-12+/AX-12A </kbd> → control table

Here I will give you some hints.
Try this:
1. use write2ByteTxRx function to change the value of Control table 6 (CW position limit) and 8 (CCW position limit) into 0.
2. The readwrite example might be showing how to change the 2byte values by using the function write2ByteTxRx, so make your application by using this code.
3. Note that when you change the values of 6 and 8th address by 2 bytes each in the control table, the motor torque should be disabled, since those items are in ROM region.

Ask further questions after doing described.

ROBOTIS-Leon, 2017-11-24:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-11-24 by ROBOTIS-Leon

shettysameeksha, 2019-01-18:
could anybody please let me know.
How to shift from wheel mode to servo mode in ax12a

kijongGil, 2019-01-18:
Hi @shettysameeksha :)
You can set easily AX-12A to the wheel mode using [R+](http://emanual.robotis.com/docs/en/software/rplus2/manager/) 
Refer to AX-12A [control table](http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/) and set 0 to CW Angle Limit and CCW Angle Limit.

shettysameeksha, 2019-01-19:
Hi @kijongGil 
Thank you.
It helped me :+1:

nitinsonker78, 2020-09-05:
can anybody help me with Dynamixel XL320  for joint mode to wheel mode with arduino uno

B3n77, 2022-07-07:
Hi, I need some assistance, I have the XM540-W150R and was wondering how to set the goal positions above 4095 and below 0 I need them to continuously turn until it hits a goal position?

shiba-8ro, 2022-07-11:
@B3n77 
Hi, You can use Extended Position Control Mode(Multi-turn)
https://emanual.robotis.com/docs/en/dxl/x/xm540-w150/#operating-mode11

B3n77, 2022-07-11:
Oh ok thank you.