ROBOTIS-GIT/DynamixelSDK #130: Protocol 2 to work with LabVIEW Real Time  [help wanted]

danielrwebb, 2017-10-18:
Please fill out the questionaire to give you the best support service.  

When trying to deploy LabVIEW code onto a Real-Time module, Protocol 2 commands throw an error regarding the SetUpComm located in Kernal32.dll .  This is becuase the Phar Lap ETS OS does not fully support all windows functions.  Protocol 1 works great since it is all based off Visa commands.  I am sure that I am missing a lot of detail, but this is what I could gather.

- Which DYNAMIXEL is it?
M42-10-S260-R

- Which CONTROLLER is it?
LabVIEW Real Time

- Which VERSION of DynamixelSDK is it?
3.5.2
  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2017-10-20:
@ddrizzle 

Hi,
sorry for the late reply

the issue seems like depended on the external library. Since we have no much bandwidth for solving this issue, probably the issue bulletin in LabVIEW communities can be recommended for use. 

I will leave this issue anyway as 'help wanted'.

If you have any other ideas on anything, please leave comments here

ddrizzle, 2017-10-20:
The references are pretty empty regarding this issue in the LabVIEW
communities.  I was able to send and receive packets from the Dynamixel
Pros yesterday in Labview, but now I think its all just rewrapping.

ROBOTIS-Leon, 2017-11-24:
@ddrizzle 

Yeah, it is a rewrapped one of c ver. SDK. We aimed to provide the Dynamixel solution in simplest way in the first place. In the other hands, we are still trying to understand the LabVIEWer's needs, though not much solutions could be provided in LabVIEW wise because of its platform-dependency.

Thanks for patience,  and please give me feedbacks if you've succeeded to do something :)

ROBOTIS-Leon, 2018-04-09:
@ddrizzle

How are you,

this issue have been wanting a help from long time ago, 
but still no action is shown. I leave the question label,
but want an understanding of closing this issue. 

We will reopen the issue if any actions occur.

closed 2018-04-09 by ROBOTIS-Leon