ROBOTIS-GIT/DynamixelSDK #131: Problem with RoboPlus

GrigoriyAY, 2017-10-18:
Unfortunately RoboPlus stopped working, I cannot launch the Wizard and get an error: **an unexpected error of the external database of driver**. Early (during several weeks), RoboPlus started Wizard, but couldn't find servomotors (MX64T, MX28T and AX12) and wrote the same error. I use USB2Dynamixel.
Could you help me?

ROBOTIS-Leon, 2017-10-20:
@GrigoriyAY 

Hi,
sorry for the late reply

It seems like what you said is not directly related with DynamixelSDK talk, so next time, please contact us by contactus@robotis.com and get adequate answer.

BTW, I will let the team know about this issue.

GrigoriyAY, 2017-10-24:
Do you know, it is possible to return Dynamixel initial settings in MATLAB?

ROBOTIS-Leon, 2017-10-24:
@GrigoriyAY 

Hi, again

To get Dynamixel initial settings manually, use read2ByteTxRx functions (in your case with MX series; MX series in Protocol 2.0 mode uses some 4 byte items in such a present position, goal position, etc., so then use read4ByteTxRx ).

read all the values from each control table address of the Dynamixel.

MATLAB uses C library basically, so you might be understood if you see some examples written in C language, and compare with the examples made with MATLAB.

In C, it should be coded like,

dxl_model_number = read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_MODEL_NUMBER (you can find the number in http://support.robotis.com/))
dxl_version_of_firmware = read1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_VERSION_OF_FIRMWARE)

and so on...

so, in MATLAB, the function boxes should be placed for every reading of current Dynamixel informations

ROBOTIS-Leon, 2017-10-24:
@GrigoriyAY 

and here is the news.

Since there was an update of Microsoft recently [LINK here](https://social.technet.microsoft.com/Forums/en-US/45ebe780-967e-4748-a146-9ec7a15626dc/unexpected-error-from-external-database-driver-1?forum=itmanager), it seems like not only our software but also the others are suffering from it. 

To solve this, as I heard from the team, 
1. disable the windows automatic update

2. Go to control panel - Program - Program and functions - update installation
then, delete Microsoft Windows (KB404****) which is installed in 11~12th October (the file name started with KB404... might be different depends on the computer)

Please try this, then give me feedbacks

Thanks

ROBOTIS-Leon, 2017-10-24:
@GrigoriyAY 

Check this

http://en.robotis.com/BlueAD/board.php?bbs_id=news&mode=view&bbs_no=1140102&page=1&key=&keyword=

GrigoriyAY, 2017-10-24:
**@ROBOTIS-Leon**

Thank you very much! Problem with RoboPlus now is solved. It was realy due to  windows automatic update.

And also for answer on the question about initial conditions, I can implement it using command for each feature.

ROBOTIS-Leon, 2017-11-24:
@GrigoriyAY 

Great

closed 2017-11-24 by ROBOTIS-Leon