ROBOTIS-GIT/DynamixelSDK #134: Slow serial read/write in Matlab

xzhang88, 2017-10-20:
Currently any read/write command using the SDK is taking ~15ms even at 1M baud rate. Is there a way to speed things up?

Using:
  [MX-64A;MX-28AT]
  [PC via USB2Dynamixel] 
 [Matlab r2016b]
  [SDK 3.4.5]

danielasun, 2017-10-21:
Have you tried lowering the latency settings for the COM port? The default is 16 ms, but you can lower it to 1ms.

xzhang88, 2017-10-24:
Yes that reduced my ping time to about 2ms per servo per read/write. But is still too slow if I need to read/write 6 servos each time. Is there a way to bypass the windows limit or would I need to implement a dedicated lower level control board?

danielasun, 2017-10-24:
2 ms per instruction is just about the limit. Have you tried using the syncwrite and syncread functions? They can write to multiple motors at once. If you need to read/write to multiple things at the same time, try using bulk read/write.

xzhang88, 2017-10-25:
Whats the difference between the two? Syncwrite/read vs bulk read/write? Based on your reply they sound to be the same. Is the bulk read/write sequential/buffered?

danielasun, 2017-10-26:
The bulk read and write commands can be used even if the Address and Data
Length of the data for each device are not all the same. You can reference
the online instruction/status packet reference here:
http://support.robotis.com/en/product/actuator/dynamixel_pro/communication/instruction_status_packet.htm

xzhang88, 2017-10-27:
I got the ping rate to ~8ms for pinging 6 servos using the sync write on Protocol 2. The problem is that I need to control end-point position AND velocity to keep a smooth trajectory (or it jerks like crazy with pos control, even after PID tuning). Unlike the DXL_Pros, the MX servos only lets me control either desired pos. or vel. and not both in the controller. I can switch to torque control and build my own feedback loop but ~8ms loop rate is too slow for that... 

So I'm guessing I need to use to a lower level controller that can take advantage of the 1M baudrate and not be bottle necked by the Windows latency. Do you have any recommendations that works with the SDK?

danielasun, 2017-10-27:
Have you tried using a different trajectory profile? Linearly interpolation between points  can be very jerky. You might try using a cubic, quartic or cycloid acceleration profile to smoothly interpolate between key frames. 
I'd recommend trying that over using velocity/torque control.

xzhang88, 2017-10-27:
I have a smoothed and densely sampled input position vector for each servo that's calculated from the inverse kinematics of the desired endpoint trajectory. Currently, I sync write the pos command to all servos, wait until the end-effector is at the desired point, then write the next pos command. Problem is: the arm will start/stop for each end point, making it jerky. I can limit the acceleration profile in protocol 2 which helps but slows down the whole thing to a crawl.

ncalvr, 2017-11-12:
I find the acceleration profile is really skewed down the low end in protocol 2.0. Even going up from 70 - 200 can make a big difference in speed but still reduce that jerkiness you talk about.

ROBOTIS-Leon, 2017-11-28:
@xzhang88 

Hi guys,
What a good you are talking about.

I think that you can get more information while looking in manipulation wise from :
https://github.com/ROBOTIS-GIT/ROBOTIS-MANIPULATOR-H/issues
Actually, that repo is dealing with those kinda story.

and the first question you had asked was completely as @danielasun  said, so keep going if you don't have more doubt in communication speed wise.

and, @ncalvr has a right view point of each instruction's characteristics.

Anyway, if you have more question, I think that you can get the solution from :
http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna
, since this seems like not only related with the SDK.

Thanks, and I will close this issue if here leaves idle until next week.

ROBOTIS-Leon, 2018-04-09:
@xzhang88 @danielasun

How are you,

this issue have been wanting a help from long time ago,
but still no action is shown. I leave the question label,
but want an understanding of closing this issue.

We will reopen the issue if any actions occur.

closed 2018-04-09 by ROBOTIS-Leon

iamstrang3, 2024-05-29:
Is there a solution to this?