ROBOTIS-GIT/DynamixelSDK #136: dynamixel_sdk is failing to build in the Kinetic debian Jessie release

tfoote, 2017-10-27:
http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__dynamixel_sdk__debian_jessie_amd64__binary/

http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__dynamixel_sdk__debian_jessie_amd64__binary/18/console
```
23:05:18 dh_auto_configure -- \
23:05:18 	-DCATKIN_BUILD_BINARY_PACKAGE="1" \
23:05:18 	-DCMAKE_INSTALL_PREFIX="/opt/ros/kinetic" \
23:05:18 	-DCMAKE_PREFIX_PATH="/opt/ros/kinetic"
23:05:18 	mkdir -p obj-x86_64-linux-gnu
23:05:18 	cd obj-x86_64-linux-gnu
23:05:18 	cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_PREFIX_PATH=/opt/ros/kinetic
23:05:18 -- The C compiler identification is GNU 4.9.2
23:05:18 -- The CXX compiler identification is GNU 4.9.2
23:05:18 -- Check for working C compiler: /usr/lib/ccache/cc
23:05:19 -- Check for working C compiler: /usr/lib/ccache/cc -- works
23:05:19 -- Detecting C compiler ABI info
23:05:19 -- Detecting C compiler ABI info - done
23:05:19 -- Check for working CXX compiler: /usr/lib/ccache/c++
23:05:19 -- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
23:05:19 -- Detecting CXX compiler ABI info
23:05:19 -- Detecting CXX compiler ABI info - done
23:05:19 -- Using CATKIN_DEVEL_PREFIX: /tmp/binarydeb/ros-kinetic-dynamixel-sdk-3.5.2/obj-x86_64-linux-gnu/devel
23:05:19 -- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
23:05:19 -- This workspace overlays: /opt/ros/kinetic
23:05:19 -- Found PythonInterp: /usr/bin/python (found version "2.7.9") 
23:05:19 -- Using PYTHON_EXECUTABLE: /usr/bin/python
23:05:19 -- Using Debian Python package layout
23:05:19 -- Using empy: /usr/bin/empy
23:05:19 -- Using CATKIN_ENABLE_TESTING: ON
23:05:19 -- Skip enable_testing() when building binary package
23:05:19 -- Using CATKIN_TEST_RESULTS_DIR: /tmp/binarydeb/ros-kinetic-dynamixel-sdk-3.5.2/obj-x86_64-linux-gnu/test_results
23:05:19 -- Looking for include file pthread.h
23:05:19 -- Looking for include file pthread.h - found
23:05:19 -- Looking for pthread_create
23:05:19 -- Looking for pthread_create - not found
23:05:19 -- Looking for pthread_create in pthreads
23:05:19 -- Looking for pthread_create in pthreads - not found
23:05:19 -- Looking for pthread_create in pthread
23:05:19 -- Looking for pthread_create in pthread - found
23:05:19 -- Found Threads: TRUE  
23:05:19 -- Found gtest sources under '/usr/src/gtest': gtests will be built
23:05:19 -- Using Python nosetests: /usr/bin/nosetests-2.7
23:05:19 -- catkin 0.7.6
23:05:20 CMake Error at CMakeLists.txt:51 (target_sources):
23:05:20   Unknown CMake command "target_sources".
23:05:20 
23:05:20 
23:05:20 -- Configuring incomplete, errors occurred!
23:05:20 See also "/tmp/binarydeb/ros-kinetic-dynamixel-sdk-3.5.2/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
23:05:20 See also "/tmp/binarydeb/ros-kinetic-dynamixel-sdk-3.5.2/obj-x86_64-linux-gnu/CMakeFiles/CMakeError.log".
23:05:20 dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_PREFIX_PATH=/opt/ros/kinetic returned exit code 1
23:05:20 debian/rules:26: recipe for target 'override_dh_auto_configure' failed
23:05:20 make[1]: *** [override_dh_auto_configure] Error 2
23:05:20 make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-dynamixel-sdk-3.5.2'
23:05:20 debian/rules:23: recipe for target 'binary' failed
23:05:20 make: *** [binary] Error 2
23:05:20 dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
23:05:20 E: Building failed
23:05:20 Traceback (most recent call last):
23:05:20   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
23:05:20     subprocess.check_call(cmd, cwd=source_dir)
23:05:20   File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
23:05:20     raise CalledProcessError(retcode, cmd)
23:05:20 # END SUBSECTION
23:05:20 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-kinetic-dynamixel-sdk']' returned non-zero exit status 1
23:05:20 
23:05:20 ------
```

Same problem on arm64: http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__dynamixel_sdk__debian_jessie_arm64__binary/21/console

civ0, 2017-10-27:
This error was introduced in https://github.com/ROBOTIS-GIT/DynamixelSDK/commit/d85b44e762ef4375a80f855ef1a3da5e354333f5#diff-d0c985ad85d1d82ffa2983f1ccc983d0

The usage of target_sources() in DynamixelSDK/c++/CMakeLists.txt causes this error, because it requires CMake >= 3.1, but Debian Jessie only has CMake 3.0.2.

tfoote, 2017-10-27:
If rolling back to an older CMake usage is not possible. You can blacklist the package here: https://github.com/ros-infrastructure/ros_buildfarm_config/blob/production/kinetic/release-jessie-build.yaml#L14 and https://github.com/ros-infrastructure/ros_buildfarm_config/blob/production/kinetic/release-jessie-arm64-build.yaml#L14 so it will stop trying to build. 

Note that I'd recommend updating the minimum cmake version declaration to get a better error message: https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/CMakeLists.txt#L4

ROBOTIS-Leon, 2017-10-30:
@tfoote @civ0 

Hi, guys
I handled the root of the issue 
the function which occurred the error but is depended on the CMake version
is replaced into good one.

I think that it will run in the RPi or some other platforms, but let me know if it is not going to do work well

Cheers

closed 2017-11-24 by ROBOTIS-Leon