ROBOTIS-GIT/DynamixelSDK #145: Issues with finding dynamixel

hozhu, 2017-11-09:
Dear Robotis Representative,

My team recently purchased a turtlebot3 from robotis and the package came with a intel joule 570x board and OpenCR 1.0 board. We were trying to control one DYNAMIXEL XM430-W210-R by using the intel joule 570x board and the OpenCR 1.0 board. The program that we used was the simplest Dynamixel tutorial from the following website (http://wiki.ros.org/dynamixel_workbench_single_manager/Tutorials/CommandLine). By following the instruction, we were able to complete step 1, step 2 and step 3 without any issues. However, when we were executing a command line from step 4 (**roslaunch my_dynamixel_workbench_tutorial single_manager.launch**) ,  we received the following error: **Please Check USB Port authorization and Baudrate [ex: 57600,115200, 1000000, 3000000] ...Failed to find dynamixel!**. We have double checked the baudrate and model ID and they seemed to be correct.  We can't figure out why it is failing to find dynamixel. To make things clearer, we are attaching an image the entire process. Thank you for your help and time in advance! 

SIncerely,

Hong
[image]

ROBOTIS-Leon, 2017-11-24:
@hozhu 

Hi, 
and sorry for the late. I hope that still you are working on this

To summarize, you should change the firmware of the OpenCR from the TurtleBot3 Firmware into signal converter firmware. 

signal converter-like firmware will be placed (today or next week) in following link:

https://github.com/ROBOTIS-GIT/OpenCR/tree/develop/arduino/opencr_arduino/opencr/libraries/turtlebot3

What is shown in the screen you attached shows that you want to use the OpenCR as USB2Dynamixel or some device that can change the usb serial to TTL level communication signal. To use DynamixelSDK example or the applications in the PC, you should change the firmware of the OpenCR as it is able to work like USB2Dynamixel. Currently the OpenCR must be working as TurtleBot3 motordriver but not the USB2Dynamixel. 

Thanks for patience, and please give me feedbacks

closed 2017-12-01 by ROBOTIS-Leon

ROBOTIS-Leon, 2017-12-06:
This issue is moved to:

https://github.com/ROBOTIS-GIT/turtlebot3/issues/90