ROBOTIS-GIT/DynamixelSDK #146: weird performance of 'Read' between Ubuntu 14.04 and Ubuntu 16.04

HyHuang1995, 2017-11-16:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　MX-64AT　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　PC　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　latest　　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

Hi,

I have tried the latest sdk on Ubuntu 16.04, but I found a bad performance of simple 'Read' function that it takes 0.19s to read 2 bytes data (e.g. current position) from 12 servos. While it turns out only 0.011s to take on Ubuntu 14.04. By the way, both baud rates are 1M.

I have checked [Issue#59](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/59). But it doesn't help with this issue. So I am wondering is there anything I need to configure on Ubuntu 16.04 to boost up USB read performance? : )

Below is my sample code.
 
```
#include "dynamixel_sdk.h"
#include "dynamixel_enum.h"

#include <iostream>
#include <chrono>

#define DEVICENAME                      "/dev/ttyUSB0"
#define PROTOCOL_VERSION                1.0
#define BAUDRATE                        1000000

using namespace std;

int main(void)
{
    dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);
    dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);
    dynamixel::GroupBulkRead groupBulkRead(portHandler, packetHandler);

    std::vector<uint8_t> id;
    uint8_t error = 0;
    int comm_res = COMM_TX_FAIL;
    std::vector<uint16_t> position;

    id.push_back(11); id.push_back(10); id.push_back(9);
    id.push_back(12); id.push_back(13); id.push_back(14);
    id.push_back(4); id.push_back(3); id.push_back(2);
    id.push_back(5); id.push_back(6); id.push_back(7);
    position.resize(id.size());

    // Open port
    portHandler->openPort();

    // Set port baudrate
    portHandler->setBaudRate(BAUDRATE);

    chrono::steady_clock::time_point start, end;
    for (auto &i : id)
        packetHandler->write1ByteTxOnly(portHandler, i, ADDR_MX_TORQUE_ENABLE, 0);
    
    while (true)
    {
        start = chrono::steady_clock::now();
        for (int i = 0; i < id.size(); ++i) {
            uint16_t pos_ = 0;
            comm_res = packetHandler->read2ByteTxRx(portHandler, id[i], ADDR_MX_PRESENT_POSITION, &pos_, &error);

            position[i] = pos_;
        }
        end = chrono::steady_clock::now();
        chrono::duration<double> duration = chrono::duration_cast<chrono::duration<double>>(end - start);
        cout << duration.count() << endl;

        for (auto &p : position)
            std::cout << ' ' << (int)p;
        std::cout << std::endl;
    }

    return 0;
}

```

ROBOTIS-Leon, 2017-11-27:
@HyHuang1995 

Hi,
and sorry for the late.

I think that this issue occurs due to #116 

Please do 

$ echo 1 | sudo tee /sys/bus/usb-serial/devices/ttyUSB0/latency_timer

and modify the code
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp
Line 47

from 8 to 2 (or appropriate number for your PC, but it will do faster communication in lower number. 

OR, try use the source uploaded in develop branch

Thanks and please give me feedbacks

HyHuang1995, 2017-11-27:
Hi, Leon,

Thanks for your reply. By
`$ echo 1 | sudo tee /sys/bus/usb-serial/devices/ttyUSB0/latency_timer`
the lantency problem is do solved.

I think I might try code in develop to see whether it has a better communication performance.

Again thank you for your help.

astridpetitjean, 2017-12-13:
Hi, 

I am also working on Ubuntu 16.04 and a MX64AR motor, communicating via USB2Dynamixel. Using
`$ echo 1 | sudo tee /sys/bus/usb-serial/devices/ttyUSB0/latency_timer`
indeed improved the performance (from 15Hz to 50Hz), however it is still too slow, I need AT LEAST 100Hz, which should be possible - as colleagues are commanding the motors without any problems at 500Hz (read & write).
Does anyone have an idea how to solve this problem? 

Thanks!

ROBOTIS-Leon, 2017-12-15:
@astridpetitjean 

Hi,
there are some ways actually.. the API, `readNByteTxRx`, for instance, is made up as `readNByteTx` and `readNByteRx`. `readNByteTxRx` transmits the data via `readNByteTx` and receives the data via `readNByteRx`. The problem is that the `readNByteRx` waits until the returning packet ordered from `readNByteTx`, and during those terms, processor can't do anything. 

So if you want to make the comm faster, put the function Tx and Rx seperately, and do like
Tx 1st
Tx 2nd
...
Rx 1st
Rx 2nd

HyHuang1995, 2017-12-15:
@ROBOTIS-Leon 

Will this method bring some trouble like when Tx the 2nd packet the bus is used by the 1st return packet?

ROBOTIS-Leon, 2017-12-15:
@HyHuang1995 

Hi, 
good question.
See:
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/e4ec5da6f6f61b0b9a7c52634d9996b122fdbe0b/c%2B%2B/src/dynamixel_sdk/protocol2_packet_handler.cpp#L670

guess what does make another problem? how'd you think

closed 2017-12-28 by ROBOTIS-Leon