ROBOTIS-GIT/DynamixelSDK #148: Problems with MX-28 with Protocol 2.0

Alienmaster, 2017-11-24:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it? [MX-28]

- Which CONTROLLER is it? [PC via USB2Dynamixel]

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [RoboManager Plus 2.0]

I have got a problem in general with the new Protocol and didn't find a better place to ask the question - sorry.

We updated three MX-28 and one MX-64 with the the newest Firmware 41 and got a problem with the MX-28 with protocol 2.0 . The MX-64 work fine but all three MX-28 dont move (with the old protocol they worked fine! We tested it before). You can connect them via USB2Serial and read out the actual position and change them but when you try to manipulate the degrees they got an error "overload". It happens in different Operation Modes. If you change while running the software the MX-28 with a MX-64 and try to move the motor works fine so it isnt a problem with the R+ software i think.

I hope you can understand my problem m)
sorry for my bad bad english

ROBOTIS-Leon, 2017-11-27:
@Alienmaster 

Hi,
and I'm sorry for the inconvenience.

We are currently tracking on the issue you commented, and the response will be in a week.

The issue which is related with the things except DynamixelSDK is able to be asked via :
http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

Thank you for your patience

ROBOTIS-Leon, 2017-12-01:
@Alienmaster 

Hi,
We did some test as you described, and couldn't find any problem occurred as what you are watching.

I should know now about :
1. QC number written on the sticker that is attached on a side of MX-28
2. the number of the firmware that was used before you updated the new one

Alienmaster, 2017-12-06:
Hi,
the QC number isn't readable but i would guess its "132C". The stickers are oily so its more guessing than reading. Some of our readable QC numbers (from different MX-28 motors) are 1202 and 1128 for example (They could were bought together but i don't know)

I tried to move the motors via the SDK Python Script on Linux "read_write.py"

When i first try to ping them with ping.py - works

[abc@abc protocol2_0]$ sudo python ping.py
Succeeded to open the port!
Succeeded to change the baudrate!
[ID:001] ping Succeeded. Dynamixel model number : 30

then i tried to move them with read_write.py. First of all i changed the DXL_MINIMUM_POSITION_VALUE and MAXIMUM Value to 0 and 4028 (see https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/4.2.2.1-CPP-protocol2.0-read_write.cpp)

My terminal says: 
[abc@abc protocol2_0]$ sudo python read_write.py 
Succeeded to open the port!
Succeeded to change the baudrate!
[RxPacketError] The data value is out of range!
Press any key to continue! (or press ESC to quit!)
[RxPacketError] The data value is out of range!
[ID:001] GoalPos:000  PresPos:-1945990398
[ID:001] GoalPos:000  PresPos:-1945990398
[ID:001] GoalPos:000  PresPos:-1945990398
[ID:001] GoalPos:000  PresPos:-1945990398
[ID:001] GoalPos:000  PresPos:-1945990398

and so on. That happens with different motors.

Another try with rebooting.py gets me this and a flashing LED:
[abc@abc protocol2_0]$ sudo python rebooting.py 
Succeeded to open the port!
Succeeded to change the baudrate!
Press any key to reboot
See the Dynamixel LED flickering
[ID:001] reboot Succeeded

So the connecting is working but the Values of the PresPos are totally confusing. Both motors react the same way.
First they were updated to Version 40 and later to 41 with no difference. I don't know the old Firmware version but a colleague said they weren't updated anytime before.

I hope you can get a better view on the problem with the new informations.

Thanks in advance,
Robert

ROBOTIS-Leon, 2017-12-07:
@Alienmaster 

Hi, again,

``` bash
the QC number isn't readable but i would guess its "132C". The stickers are oily so its more guessing than reading. Some of our readable QC numbers (from different MX-28 motors) are 1202 and 1128 for example (They could were bought together but i don't know)
```
okay, it looks a little bit old.

``` bash
I tried to move the motors via the SDK Python Script on Linux "read_write.py"

When i first try to ping them with ping.py - works

[abc@abc protocol2_0]$ sudo python ping.py
Succeeded to open the port!
Succeeded to change the baudrate!
[ID:001] ping Succeeded. Dynamixel model number : 30

then i tried to move them with read_write.py. 

First of all i changed the DXL_MINIMUM_POSITION_VALUE and MAXIMUM Value to 0 and 4028 

(see https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/4.2.2.1-CPP-protocol2.0-read_write.cpp)
```

the Maximum Position value of MX-28 is 2047. See here : http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-28at_ar.htm

if you give bigger than 2047, it wouldn't move.

the message
`[RxPacketError] The data value is out of range!` means that you gave a bigger value so that it couldn't run as ordered.
 
Let's change it first.

Alienmaster, 2017-12-13:
I changed the Range to 0-2047 see:
DXL_MINIMUM_POSITION_VALUE  = 0
DXL_MAXIMUM_POSITION_VALUE  = 2047

and the result is the same:
Succeeded to open the port!
Succeeded to change the baudrate!
b'[RxPacketError] The data value is out of range!'
Press any key to continue! (or press ESC to quit!)
b'[RxPacketError] The data value is out of range!'
[ID:001] GoalPos:000  PresPos:-1945990398

Im confused about the PresPos - i thought it should be in a Range about 0-2047 and not a negativ billion 945 million...

ROBOTIS-Leon, 2017-12-14:
@Alienmaster 

Hi,

if the printed message starts with `b'`, it means that the SDK you've downloaded is old version.
It was solved in recent, so check which version you've get. 

The newest is 3.5.4.

ROBOTIS-Leon, 2017-12-28:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-12-28 by ROBOTIS-Leon

yoneken, 2018-03-14:
Hi, I think I got an exactly the same issue.

Which DYNAMIXEL is it? [XM430-W350-R]
Which CONTROLLER is it? [Ubuntu PC via U2D2]
Which VERSION of DynamixelSDK is it? [3.5.4]

> yoneken@bundle:~/ros/src/dynamixel_sdk/python/protocol2_0$ python ./read_write.py 
> Succeeded to open the port!
> Succeeded to change the baudrate!
> [RxPacketError] The data value is out of range!
> Press any key to continue! (or press ESC to quit!)
> [RxPacketError] The data value is out of range!
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398
> [ID:001] GoalPos:200  PresPos:-1945990398

baertscc, 2018-03-26:
Hi there, I've got the exact same problem as well:

Which DYNAMIXEL is it? [XM540-W270-R]
Which CONTROLLER is it? [PC via USB2Dynamixel]
Which VERSION of DynamixelSDK is it? [3.5.4 - C++]

> Succeeded to open the port!
> Succeeded to change the baudrate!
> [RxPacketError] The data value is out of range!
> Press any key to continue! (or press ESC to quit!)
> [RxPacketError] The data value is out [...]

When using the R+ Manager, the Dynamixel works fine and all Rx Packages contain the correct data. When using the SDK, it always leads to this bug.
The error is consistently reproducible under Windows (Win 10) and Linux (Ubuntu 16.04).

ahmaddidiks, 2020-09-02:
i've tried change DXL_MINIMUM_POSITION_VALUE MAXIMUM 0 - 2047 too, but I also got this error
Succeeded to open the port!
Succeeded to change the baudrate!
[RxPacketError] The data value is out of range!
Press any key to continue! (or press ESC to quit!)