ROBOTIS-GIT/DynamixelSDK #152: [3.5.4][Python][USB2Dynamixel2][MX-64&MX-106]Issue reading present temperature

alborzs, 2017-12-06:
Hi,
According to the documentation of both **MX-64** and **MX-106** the **present temperature** address is at **146** and it should be **1 Byte**.
When performing a Read of 1 Byte on that address I seem to be getting random numbers such as 1739039.
But when I perform a 2 Byte Read I get correct temperature reading.
I have attached the code below. Either your documentation is wrong (temperature is in fact 2 Byte when it says 1 Byte in documentation) or there is something else wrong in the SDK.

I am using Protocol version 2.0 and I have updated my servos to the latest version of firmware just now and I still have the same issue.

```
ADDR_PRESENT_TEMPERATURE            = 146      # 1 Byte

dxl_present_temperature = dynamixel.read2ByteTxRx(port_num, PROTOCOL_VERSION, dynomixel_id, ADDR_PRESENT_TEMPERATURE)
    comm_result = dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION)
    comm_error = dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION)
    if comm_result != COMM_SUCCESS:
        print(dynamixel.getTxRxResult(PROTOCOL_VERSION, comm_result))
    elif comm_error != 0:
        print(dynamixel.getRxPacketError(PROTOCOL_VERSION, comm_error))

    print("[ID:%03d] Temperature: %02d" % (dynomixel_id, dxl_present_temperature))
```

ROBOTIS-Leon, 2017-12-07:
@alborzs 

Hi,

There was no problem on getting the Dynamixel temperature from following environment:

Motor model: MX-64  (QC @ 1640)
Motor firmware: 41
SDK ver : 3.5.4
Baudrate : 57,600 bps

Test code:

``` py
#!/usr/bin/env python
# -*- coding: utf-8 -*-

# Author: Ryu Woon Jung (Leon)

import os

if os.name == 'nt':
    import msvcrt
    def getch():
        return msvcrt.getch().decode()
else:
    import sys, tty, termios
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)
    def getch():
        try:
            tty.setraw(sys.stdin.fileno())
            ch = sys.stdin.read(1)
        finally:
            termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)
        return ch

os.sys.path.append('../dynamixel_functions_py')             # Path setting

import dynamixel_functions as dynamixel                     # Uses Dynamixel SDK library

# Control table address
ADDR_PRO_TORQUE_ENABLE       = 64                          # Control table address is different in Dynamixel model
ADDR_PRO_PRESENT_TEMPERATURE = 146

# Protocol version
PROTOCOL_VERSION            = 2                             # See which protocol version is used in the Dynamixel

# Default setting
DXL_ID                      = 1                             # Dynamixel ID: 1
BAUDRATE                    = 57600
DEVICENAME                  = "/dev/ttyUSB0".encode('utf-8')# Check which port is being used on your controller
                                                            # ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

TORQUE_ENABLE               = 1                             # Value for enabling the torque
TORQUE_DISABLE              = 0                             # Value for disabling the torque

ESC_ASCII_VALUE             = 0x1b

COMM_SUCCESS                = 0                             # Communication Success result value
COMM_TX_FAIL                = -1001                         # Communication Tx Failed

# Initialize PortHandler Structs
# Set the port path
# Get methods and members of PortHandlerLinux or PortHandlerWindows
port_num = dynamixel.portHandler(DEVICENAME)

# Initialize PacketHandler Structs
dynamixel.packetHandler()

dxl_comm_result = COMM_TX_FAIL                              # Communication result

dxl_error = 0                                               # Dynamixel error

# Open port
if dynamixel.openPort(port_num):
    print("Succeeded to open the port!")
else:
    print("Failed to open the port!")
    print("Press any key to terminate...")
    getch()
    quit()

# Set port baudrate
if dynamixel.setBaudRate(port_num, BAUDRATE):
    print("Succeeded to change the baudrate!")
else:
    print("Failed to change the baudrate!")
    print("Press any key to terminate...")
    getch()
    quit()

# Read present position
dxl_present_temp = dynamixel.read1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_PRESENT_TEMPERATURE)
dxl_comm_result = dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION)
dxl_error = dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION)
if dxl_comm_result != COMM_SUCCESS:
    print(dynamixel.getTxRxResult(PROTOCOL_VERSION, dxl_comm_result))
elif dxl_error != 0:
    print(dynamixel.getRxPacketError(PROTOCOL_VERSION, dxl_error))

print("[ID:%03d] PresTemp:%03d" % (DXL_ID, dxl_present_temp))

# Close port
dynamixel.closePort(port_num)

```

questions are welcomed

alborzs, 2017-12-08:
@ROBOTIS-Leon 
Thanks for your reply, and I tested your code (which is pretty much what I had already tested).
And the result is still a failure.
When I run your code on MX-64 I get the following:

```
Succeeded to open the port!
Succeeded to change the baudrate!

[ID:002] PresTemp:5465109
```

and when I test it on my MX-106 I get similar result.
```
Succeeded to open the port!
Succeeded to change the baudrate!

[ID:003] PresTemp:6775829
```
As explained in the title, both servos are upgraded to the latest firmware.
But if I change **read1ByteTxRx** to **read2ByteTxRx**, then I get correct result which is usually something around 21 to 31.

ROBOTIS-Leon, 2017-12-12:
@alborzs 

Hi,
to debug this, please do compare these with your environment. I think that it causes by some environmental differences.
(It is hard to debug since the problem isn't seen here)

1. Try another languages, such as c, c++
2. Try 32 bit example, if you've done the 64 bit example
3. Try another OS

, so that we can adjust the focus

My test environment is as follows.

Ubuntu 16.04.02 64bit 
Python ver 2.7.12
Dynamixel QC number(Sticker attached): 1640

closed 2017-12-28 by ROBOTIS-Leon