ROBOTIS-GIT/DynamixelSDK #154: Having trouble using the Dynamixel MX 64T & 28T with labview

tejes05, 2018-01-04:
I am having trouble using the Dynamixel MX 64T & 28T  with LabVIEW. I have found some examples but they don't have any examples of how to move all the servos at the same time ( joint mode). 

when I use Dynamixel API in labview which works for AX-12 very well, restricts the rotation up to 90deg. 

Does anyone have any examples for me to move MX-64T/ MX28T servo?
 

 Thanks for any help,

ROBOTIS-Leon, 2018-01-15:
@tejes05 

Hi, 

MX series has almost the same control table as AX series has, unless you used the R+ manager 2.0, and changed MX's firmware to the newest which makes it be controlled by protocol 2.0. If you did, it's control table must be changed a lot from the previous.

Check control tables of each of them.

AX 28 control table : http://emanual.robotis.com/docs/en/dxl/ax/ax-12w/
MX 28 control table (old firmware) : http://emanual.robotis.com/docs/en/dxl/mx/mx-28/
MX 28 control table (new firmware) : http://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/
MX 64 control table (old firmware) : http://emanual.robotis.com/docs/en/dxl/mx/mx-64/
MX 64 control table (new firmware) : http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/

tejes05, 2018-01-15:
Dear Leon,
Thanks for the reply.
the control table does not issue for me. I am new to LabVIEW and Dynamixel servo.  I am trying to move MX-64T and 28T motors in LabVIEW with help of Dynamixel sdk examples.
Can you tell me how to separately move the motors using LabVIEW examples as included in Dynamixel sdk 3.5.4 (DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\labview\protocol1.0)
I have tried to move both motors individually by making small changes in Sync_write program (DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\labview\protocol1.0\sync_write) but it fail.
Please tell me what corrections need to in programming to move both motors individually.

johncgh22, 2018-01-16:
@tejes05 
Hi
I´m been working with Dynamixel servos for a little bit, so let me try to aid you. The motors I´m using are: **AX-18A, MX-64 and MX-28**

The examples inside DynamixelSDK works fine with all this motors. If you want to move them in a single VI, just need to modify them a little: I´m gonna take the **read_write.vi** as an example. 

This VI handles each process as a sequence (Intialize, Port Open, Torque Enable, Dynamixel Control, Torque Disable, Port Close) and uses Local Variables for sending the correct numbers for each VI necesities (Dynamixel ID, Protocol Version, Address, etc). In order to make move "n" number of motors, you need to add the necessary info for each motor. As you can see, the things you need to add for each one are:

- Dynamixel ID
- Protocol Version
- Address for Pressent Position and Goal Position
- write2Byte Tx Bx.vi
- One knob to make the movement
- read2Byte Tx Bx..vi
- If you want to add more (ex. Velocity, Temperature, etc), you need to see each motor Control Table for its value, then make the control for it, then the VI it needs in orther to make it work.

This can also apply for the **sync_write.vi**, you just need to check what it needs, and add the necessary VI and info for each motors.

Hopes this help you understand how it works, and try to replicate it in your VI; and sorry if I made some gramar errors.

ROBOTIS-Leon, 2018-01-17:
@tejes05 

Hi,

Please do as @johncgh22 described, and give us feedbacks.
Thanks, @johncgh22

closed 2018-01-25 by ROBOTIS-Leon