ROBOTIS-GIT/DynamixelSDK #155: visual basic 2017 fails to build library

ZHUIA, 2018-01-13:
[3.5.4][MATLAB2017a][XM430-W350-R]
I am using the XM430-W350-R on windows 10 and MATLAB 2017a.  I'm using `DynamixelSDK ver. 3.5.4` Following the  software preparation for  matlab using the youtube tutorial. need to download visual basic and build a library on it.. so i downloaded visual basic 2017 and when i  follow the tutorial the visual basic fails to  build the library showing many errors like 
Error	C1083	Cannot open include file: 'group_sync_read.h': No such file or directory	
What should i do to fix this

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2018-01-15:
@ZHUIA 

Hi,

`visual basic 2017` ?, no it's visual studio 2017, and it compiles c language SDK files.
of course it can't include the `group_sync_read.h` file, and it doesn't know what it is

See the wiki [here](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/c_windows/#c-windows)to know how to build all things

ZHUIA, 2018-01-15:
yeah sorry VS not Vb.. fixed the issue

closed 2018-01-15 by ZHUIA