ROBOTIS-GIT/DynamixelSDK #156: General question regarding the startup procedure and calibration of MX-series

alborzs, 2018-01-15:
I don't know how to tackle this. This is not an issue, more a general question and I hope that this is the right place to ask it.

I have a couple of MX-106 and MX-64 connected and every time I power up the whole system, I request each servo to move to the 0 position (even if they already are at position 0). This makes each servo start to move and they stop at position 0, sometimes they make a whole revolution to stop at position 0.
My question here is, is there a way to make the servo move to position 0 but to take the shortest possible way there instead of making a whole revolution in case it is already very close to position 0?

ROBOTIS-Leon, 2018-01-17:
@alborzs 

Hi,

and this is the idea from ROBOTIS HQ: Yes, it can't remember where it was going after shutdown and reboot, so you should make the controller remember before it happens.

It would be a great help if someone has different ideas

ROBOTIS-Leon, 2018-04-09:
@alborzs

Hi,

to make the issue be solved, I will send this to [robotis forum](http://en.robotis.com/service/forum_view.php?slg=&page_type=&bbs_no=2406752).
you can follow up the status of how your issue is being solved.

closed 2018-04-09 by ROBOTIS-Leon