ROBOTIS-GIT/DynamixelSDK #157: problem to connect U2D2 and mx-106R in visual studio 2017

hamidhrj, 2018-01-19:
I'm using `U2D2` serial converter
- I'm using `DynamixelSDK ver. 3.5.4
- I'm using Language [C] / [C++] / [C#]
1- noting show in ping in [C] / [C++] / [C#]
https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/master/c/example/protocol2.0/ping
2- out of range MX-106R in
https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/master/c/example/protocol2.0/read_write

ROBOTIS-Leon, 2018-01-19:
@hamidhrj 

Hi,

I couldn't extract any issue on your comment. Could you explain of the situation more?

hamidhrj, 2018-01-19:
I'm using U2D2 serial converter and MX-106R
Run DynamixelSDK ver. 3.5.4 in visual studio 2017
when using all exampel in SDK show etc error
[http://jariteh.ir/upload/uploads/1516432823.jpg](url)
[http://jariteh.ir/upload/uploads/1516353339.jpg](url)

hamidhrj, 2018-01-19:
@ROBOTIS-Leon 
 Hi
please insert exampel for MX -106 (2) series in new version SDK 
and  include for U2D2

hamidhrj, 2018-01-20:
Why is this message
Succeeded to open the port!

Succeeded to change the baudrate!

Dynamixel has been successfully connected

Press any key to continue! (or press ESC to quit!)

**`[RxPacketError] The data length does not match as expected!   
`**

[ID:002] GoalPos:4095  PresPos:1058

ROBOTIS-Leon, 2018-01-25:
@hamidhrj 

Hi,

if you've done auto-firmware-upgrade of MX106 by Robo Plus 2.0,
and if your MX106 is set to protocol 2.0 communication option,

change the ADDR in the example code into what is described in 
http://emanual.robotis.com/docs/en/dxl/mx/mx-106-2/#control-table-of-eeprom-area

hamidhrj, 2018-01-25:
@Robotis-leon
Thank you

closed 2018-01-25 by hamidhrj