ROBOTIS-GIT/DynamixelSDK #158: [keil],[arduino]: my AX12+ does not respond at all!

ar-msb, 2018-02-01:
I am using Dynamixell AX12+. I should program it using stm32f407VG with keil but the motor does not 
respond to any commands at all. I tested the circuit on ARDUINO UNO as well, but that did not work too.

we should do a velocity control with this motor. We used the following circuit for the system:
http://nascola.hatenablog.com/entry/2016/04/26/203433
[image]
we used prepared libraries for arduino code like:
http://savageelectronics.blogspot.com/2011/01/arduino-y-dynamixel-ax-12.html
but the motor does not react to any command! it just blinks once when it is powered up by 12v, 1.5A. 
I somehow feel that none of my commands reach to the motor. can anyone help where the problem might be?

ROBOTIS-Leon, 2018-02-02:
@ar-msb 

Hi again,
I just thought that you are using currently an official DynamixelSDK, but what is shown here seems like it i is not. To let you know how to use it, it at least must use the DynamixelSDK which is being released here.

and as I mentioned in the mail, it is better getting more information related with hardware stuffs in [OpenCM 9.04](https://github.com/ROBOTIS-GIT/OpenCM9.04/issues) than here

Cheers

ar-msb, 2018-02-05:
I bought a CM9.04, and it solved our problem. I would have bought it sooner if I knew how easy it is to work with it.
Thanks for your help.

closed 2018-02-05 by ar-msb

ROBOTIS-Leon, 2018-02-21:
@ar-msb 

Hope to hear your further successful works!
Cheers