ROBOTIS-GIT/DynamixelSDK #160: Reading the entire control table in a Dynamixel MX-64 AT

gauthamvasan, 2018-02-02:
I'm using `DynamixelSDK ver. 3.5.4` and `Python` with a `USB2AX` serial converter. The current DynamixelSDK code and the hardware works great. But I'm unable to figure out how to read the entire control table of one specific motor (I'm using a Dynamixel MX-64AT with Protocol 1.0). I tried using the `readTxRx` function followed by the `getReadData` function. That works perfectly when I'm reading up to 2 bytes. I looked at the C code and it looks like getReadData isn't configured to handle more than 2 bytes. 

Is there any other method call that could be used to do this? Any help would be deeply appreciated. Thanks!

gauthamvasan, 2018-02-07:
@ROBOTIS-Leon I misunderstood the getDataRead function due to my rusty knowledge of C. For some stupid reason, I expected C to return arrays like python. The current code works well for my purposes and the issue is closed from my side. Thanks for your help!

ROBOTIS-Leon, 2018-02-13:
@gauthamvasan 

Great

closed 2018-02-13 by ROBOTIS-Leon