ROBOTIS-GIT/DynamixelSDK #161: LOADLIBRARY error in MATLAB 2017a and b.. Visual Studio 2017.. Windows 10

ZHUIA, 2018-02-12:
I'm using DynamixelSDK ver. 3.5.4 and USB2Dynamixel2 serial converter for XM 430-W350-R dynamixel motor with [MATLAB 2017a] and Windows 10 (64-bit) laptop.
I followed all the instructions of building the library in VIsual studio 2017 for MATLAB, but I am having an issue when i run the Read/write sample code in MATLAB 2017a, LOADLIBRARY error is prompting up. 
For compatibility issues between VS2017 and MATLAB 2017a, I even download the patch of XML files but still the error persists. 
Tried using MATLAB 2016b, 2017b but still no success.
 If anyone has used this dynamixel with MATLAB, your suggestions will be very helpful on how to fix this issue.

Thanks

ROBOTIS-Leon, 2018-02-13:
@ZHUIA 

Hi,

Can you show me what kind of log your MATLAB shows?

and try the DynamixelSDK source in develop branch. It was a little bit modified since there was an error actually in MATLAB 2017 porting, first of all.

ZHUIA, 2018-02-13:
Thanks for the response.. what do you mean by develop branch? the matlab log is shown in the image below. 
[image]

ROBOTIS-Leon, 2018-02-19:
@ZHUIA 

Hi, and I'm sorry for my late reply.

You can get the develop branch DynamixelSDK either of ways:
1. if you have any git bash program which can run on the Windows, try the following:

``` bash
$ git clone https://github.com/ROBOTIS-GIT/DynamixelSDK.git
$ cd DynamixelSDK/
$ git checkout develop
```
then, build c library, run matlab and follow the description you saw.

2. click Button "Branch : master" and click develop to change the currently shown folder to develop.
Then, Download by clicking green button "Clone or Download" on the right side, and download it.
Finally, build c library, run matlab and follow the description you saw.

ZHUIA, 2018-02-23:
followed step 1 method on the git bash program.. but still no luck...the error is same

ROBOTIS-Leon, 2018-02-26:
@ZHUIA 

Hi, 

Is there more error code of attached?

ZHUIA, 2018-02-28:
the error is as below:

Error using loadlibrary
Building dxl_x64_c_thunk_pcwin64 failed.  Compiler output is:
C:\ProgramData\MATLAB\SupportPackages\R2017a\3P.instrset\mingw_492.instrset\bin\gcc -I"C:\Program
Files\MATLAB\R2017a\extern\include" -fexceptions -fno-omit-frame-pointer -I"C:\Users\zakia
hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\matlab\protocol2.0" -I"C:\Users\zakia
hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk" "dxl_x64_c_thunk_pcwin64.c" -o
"dxl_x64_c_thunk_pcwin64.dll" -shared
In file included from
C:/ProgramData/MATLAB/SupportPackages/R2017a/3P.instrset/mingw_492.instrset/lib/gcc/x86_64-w64-mingw32/4.9.2/include/stdint.h:9:0,
                 from C:\Users\zakia
hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk/packet_handler.h:24,
                 from C:\Users\zakia
                 hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk/group_bulk_read.h:25,
                 from C:\Users\zakia
                 hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk/dynamixel_sdk.h:24,
                 from dxl_x64_c_thunk_pcwin64.c:27:
C:/ProgramData/MATLAB/SupportPackages/R2017a/3P.instrset/mingw_492.instrset/x86_64-w64-mingw32/include/stdint.h:35:21:
error: conflicting types for 'int8_t'
 typedef signed char int8_t;
                     ^
In file included from C:\Users\zakia
hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk/dynamixel_sdk.h:23:0,
                 from dxl_x64_c_thunk_pcwin64.c:27:
C:\Users\zakia hussain\Downloads\DynamixelSDK-3.5.4\DynamixelSDK-3.5.4\c\include\dynamixel_sdk/robotis_def.h:23:29: note:
previous declaration of 'int8_t' was here
 typedef char                int8_t;
                             ^

Error in read_write (line 48)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h', 'addheader',
    'packet_handler.h');

ROBOTIS-Leon, 2018-03-05:
@ZHUIA 

Hi, again,
sorry for my late reply.

that point is what DynamixelSDK in develop branch is intending to solve.
Or, if you couldn't find the way how to use it, just modify the robotis_def.h:

``` cpp
typedef char                int8_t;
typedef short int           int16_t;
typedef int                 int32_t;

to

typedef signed char         int8_t;
typedef signed short int    int16_t;
typedef signed int          int32_t;
```

ZHUIA, 2018-03-06:
it worked... Thank you so much :)

ROBOTIS-Leon, 2018-03-06:
@ZHUIA 

Great

closed 2018-03-06 by ROBOTIS-Leon