ROBOTIS-GIT/DynamixelSDK #162: [3.5.4][C++][USB2Dynamixel][XM430-W210-T] Issue : [TxRxResult]There is no status packet

AminMohammadi, 2018-02-15:
BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1.
- I'm using `DynamixelSDK ver. 3.5.4`
- I'm using `C++` Language
- I'm using `USB2Dynamixel` serial converter
- I'm using `XM430-W210-T`
- My issue is that when I follow videos to run read_write executable file, it shows this message on my terminal:

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000

I exactly follow the video. I also checked ID number, baudrate, etc. in the code. When I run the "read_write" executable file, it gives me above message. I don't know what is the reason. I use Linux 14.04, 64bit, and my kernel version is 4.4.0-111-generic. The USB2Dynamixel converter that I use, was used in a robot arm with MX dynamixel motors before. Does it make a difference?
Any help is appreciated.

tutamc, 2018-02-16:
Can I see the code to help you better??

I don't know if this will fix the problem, but one thing that I observed the examples like read_write (protocol 2.0) is the part "Control table address", which the address it's different from the manual. You can find these address here: http://support.robotis.com/en/product/actuator/dynamixel_x/xm_series/xm430-w210.htm.

Hope this helps.

AminMohammadi, 2018-02-16:
@tutamc 
Thanks for the response. I use protocol 1.0, and here's the code(I didn't change the code, but I checked it to make sure baud rate and ID are correct):
/*******************************************************************************
#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>

#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_MX_TORQUE_ENABLE           24                  // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION           30
#define ADDR_MX_PRESENT_POSITION        36

// Protocol version
#define PROTOCOL_VERSION                1.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        57600
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      100                 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4000                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     10                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

int getch()
{
#if defined(__linux__) || defined(__APPLE__)
  struct termios oldt, newt;
  int ch;
  tcgetattr(STDIN_FILENO, &oldt);
  newt = oldt;
  newt.c_lflag &= ~(ICANON | ECHO);
  tcsetattr(STDIN_FILENO, TCSANOW, &newt);
  ch = getchar();
  tcsetattr(STDIN_FILENO, TCSANOW, &oldt);
  return ch;
#elif defined(_WIN32) || defined(_WIN64)
  return _getch();
#endif
}

int kbhit(void)
{
#if defined(__linux__) || defined(__APPLE__)
  struct termios oldt, newt;
  int ch;
  int oldf;

  tcgetattr(STDIN_FILENO, &oldt);
  newt = oldt;
  newt.c_lflag &= ~(ICANON | ECHO);
  tcsetattr(STDIN_FILENO, TCSANOW, &newt);
  oldf = fcntl(STDIN_FILENO, F_GETFL, 0);
  fcntl(STDIN_FILENO, F_SETFL, oldf | O_NONBLOCK);

  ch = getchar();
[... 112 lines cut ...]
    {
      index = 1;
    }
    else
    {
      index = 0;
    }
  }

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }

  // Close port
  portHandler->closePort();

  return 0;
}

tutamc, 2018-02-16:
Ok, now try to change the Control Table Address, like this:

24 -> 64
30 -> 116
36 -> 132

And run!

AminMohammadi, 2018-02-16:
@tutamc 
Thank you for your response.
I did that, but I still get same error. I changed the code like this:

#define ADDR_MX_TORQUE_ENABLE           64             
#define ADDR_MX_GOAL_POSITION           116
#define ADDR_MX_PRESENT_POSITION        132

I deleted executable file, and made it again. But no success.

tutamc, 2018-02-16:
Do not these phrases appear?
-----------------------------------------------------------------------
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press ESC to quit!)
-----------------------------------------------------------------------

AminMohammadi, 2018-02-16:
@tutamc 
No. Same as before. No status packet

tutamc, 2018-02-16:
Change the protocol to 2.0!!

this way: 
#define PROTOCOL_VERSION 2.0

AminMohammadi, 2018-02-16:
@tutamc 
Thanks. It worked. I got this message:

Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press ESC to quit!)

But it should work with protocol 1. Manual says that:

"Protocol 1.0 does not support addresses greater than 256. Therefore, Indirect Address 29 ~ 56 and Indirect Data 29 ~ 56 can only be accessed with Protocol 2.0."

So it should support. But the thing is that motor didn't move. I think the problem is this line:

#define DXL_MINIMUM_POSITION_VALUE      100                
#define DXL_MAXIMUM_POSITION_VALUE      4000

I'm gonna change it to see what happens. It says that motor moves if the position is between these values(It has absolute encoder not incremental). I appreciate your help.

tutamc, 2018-02-16:
Yea, I know, it's very confusing all these protocols.

Now to move the motor, you have to change another thing that I forget. 

In this line: "dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_MX_GOAL_POSITION, dxl_goal_position[index], &dxl_error);"

You have to replace "write2ByteTxRx" to "write4ByteTxRx".

AminMohammadi, 2018-02-16:
@tutamc 
I really appreciate your help. It's working now. I just have one question. How did you know that I have to change "write2ByteTxRx" to "write4ByteTxRx" in ADDR_MX_GOAL_POSITION? Why shouldn't I do that for "ADDR_MX_PRESENT_POSITION"?
Is there a library that shows all the functions and everything?

Thanks

tutamc, 2018-02-16:
You are welcome Amin...

You can check in this site: 
http://support.robotis.com/en/product/actuator/dynamixel_x/xm_series/xm430-w210.htm

You have to verify on Control Table Address, specifically, the size [byte] which "Data Name" (Address) you used.

AminMohammadi, 2018-02-16:
@ROBOTIS-Leon 
There are some mistakes in the manual. In the manual it says that "Present Position" is 4 bytes, but when I use "write4ByteTxRx" for Present Position, it gives me error. It doesn't give error for "Goal Position" though. Both of them are supposed to be 4 bytes based on manual.
Manual says that protocol 1.0 should support addresses between 57~256, but I have to use protocol 2.0 for "read_write" example.

ROBOTIS-Leon, 2018-02-17:
@tutamc 

Thank you for your kindness and great contribution

@ AminMohammadi 

Hi, 

can you show me the link? I couldn't figure out the point.

Besides, you can't control Present Position, since it is a readonly.
Present Position is a value of "currently placed position of the servo".
When you want to control the Dynamixel position, you should handle "Goal Position", not the "Present's"

Control Table Address and the memory length of each control items are the same in both Protocol 1.0 and Protocol 2.0 in X Series Dynamixel.
 
If you are mean that the source code is wrong, the source code is actually made based on Dynamixel MX
 control table address, so it varies on its series

AminMohammadi, 2018-02-19:
@ROBOTIS-Leon 
Got your message.
I have some questions:
Can you give me a link that explains what is the purpose of each code in SDK?
Is there anyway to change ID number and baudrate of my Dynamixel in Linux(I don't have Dynamixel wizard)?
Do you have a document that explains each Dynamixel command, its arguments, and everything?
I want to control my motor by torque. How accurate is that and which function should I use? Do you have a example code?
Thanks

ROBOTIS-Leon, 2018-02-19:
@AminMohammadi 

Hi, again,

```
Is there anyway to change ID number and baudrate of my Dynamixel in Linux(I don't have Dynamixel wizard)? If not, leave a link here to get the software.
```

Go [ROBOTIS.com](http://en.robotis.com/service/downloadpage.php?cate=software), and download R+ Manager 2.0, then use it. If you have any question of how to use, ask me freely, but I think that it is easy for you

```
Do you have a document that explains each Dynamixel command, its arguments, and everything?
```

Go [E-Manual](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#control-table-of-eeprom-area), seek what you want to handle

```
I want to control my motors by torque. How accurate is that and which function should I use? Do you have a example code?
```

See [Operating Mode](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#operating-mode11). In your case, you should change it into 5. Then, you should deal with the current values to control.

Dynamixel SDK doesn't have an example intended to show functionality of the Dynamixel. It shows only fundamental things. I suppose you to see, though it is programmed as one of ROS package, [Dynamixel-Workbench](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/) code and refer it.

If you have more questions, don't hesitate to ask.
Thanks

AminMohammadi, 2018-02-19:
@ROBOTIS-Leon 
Thanks for the quick response.

I saw your link. It was in windows. So there's no way to change it in Linux, right?(I have only linux on my system)

I looked at this Dynamixel-Workbench. It shows how to have access to control gains, changing addresses and so on. What I need is an example or a document that shows me how to have torque control. 

Thanks

ROBOTIS-Leon, 2018-02-19:
@AminMohammadi 

Hi, 

```
I saw your link. It was in windows. So there's no way to change ID/baudrate/etc in Linux, right?(I have only linux on my system)
```

As long as  you can deal with ROS, you can handle them in Linux ([Dynamixel Workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench) Package), but if not, there's no selection unfortunately, excepting seek connected Dynamixel by using Broadcast ping, then change them by your codes. We are sorry about that.

```
I looked at this Dynamixel-Workbench. It shows how to have access to control gains, changing addresses and so on. What I need is an example or a document that shows me how to have torque control. I mean what is the function for that?
```

There is no example that shows how to control Torque of the motor in SDK. As I told previously, it only shows how to use the API, not to show the functions of the Dynamixel. What I intended with pointing the Dynamixel-Workbench is to give you an example code of how to change the values related with the Torque control.

Some details that you may be interested with is [here](http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#performance-graph). There is no more well described one than described there

AminMohammadi, 2018-02-19:
@ROBOTIS-Leon 
I have DynamixelWizardm on my PC, but when I want to change the ID, it gives me error. Errors are overheating error and input voltage error. Power supply that I use is 12V, 5A.

ROBOTIS-Leon, 2018-02-20:
@AminMohammadi 

Great, good luck

closed 2018-02-20 by ROBOTIS-Leon

kkk123m, 2019-10-09:
Hi 

I am facing similar issue with Dynamixel SDK but servo doesn't move. Anyone know issue heere. Here is Example sketch I am  using for XL 430, with Open CM 9.04 +EXP

/*
 * Controller : OpenCM9.04 + OpenCM 485 EXP
 * Dynamixel : XL-320, X-Series, MX-Series with Protocol 2.0
 * Power source
 *   - XL-320 : LB-040/041(x2) on OpenCM9.04C
 *   - X-Series/MX-Series : SMPS 12V 5A on OpenCM 485 EXP
 * 
 * XL-320 is connected to Dynamixel TTL 3PIN of the OpenCM9.04
 * http://emanual.robotis.com/docs/en/parts/controller/opencm904/#layoutpin-map
 
 * X-Series and MX-Series are connected to Dynamixel BUS on 485 EXP board
 * http://emanual.robotis.com/docs/en/parts/controller/opencm485exp/#layout
 * 
 * This example will test only one Dynamixel at a time.
*/

#include <DynamixelSDK.h>

#define ADDRESS_XL320_TORQUE_ENABLE     24
#define ADDRESS_XL320_GOAL_POSITION     30
#define ADDRESS_XL320_PRESENT_POSITION  37

#define ADDRESS_X_TORQUE_ENABLE         64
#define ADDRESS_X_GOAL_POSITION         116
#define ADDRESS_X_PRESENT_POSITION      132

/* Uncomment below definition in order to use the sketch for X series/MX-series with Protocol 2.0 other than XL-320*/
// #define use_XL_320

#ifdef use_XL_320
  // Definition 1 : Control table address for XL-320
  // Use "Definition 1" for XL-320
  #define ADDRESS_TORQUE_ENABLE           ADDRESS_XL320_TORQUE_ENABLE
  #define ADDRESS_GOAL_POSITION           ADDRESS_XL320_GOAL_POSITION
  #define ADDRESS_PRESENT_POSITION        ADDRESS_XL320_PRESENT_POSITION
#else
  // Definition 2 : Control table address for X Series and MX Series with Protocol 2.0
  // Ues "Definition 2" for X Series or MX Series with Protocol 2.0
  #define ADDRESS_TORQUE_ENABLE           ADDRESS_X_TORQUE_ENABLE
  #define ADDRESS_GOAL_POSITION           ADDRESS_X_GOAL_POSITION
  #define ADDRESS_PRESENT_POSITION        ADDRESS_X_PRESENT_POSITION
#endif

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1

#ifdef use_XL_320
  #define BAUDRATE                        1000000             // XL-320 default baudrate : 1000000 (Default Baudrate for X series and MX series(with Protocol 2.0) is 57600)
#else
  #define BAUDRATE                        57600               // XL-320 default baudrate : 1000000 (Default Baudrate for X series and MX series(with Protocol 2.0) is 57600)
#endif

#define DEVICENAME                      "3"                 // Check which port is being used on your controller
                                                            // DEVICENAME "1" -> Serial1
                                                            // DEVICENAME "2" -> Serial2
                                                            // DEVICENAME "3" -> Serial3(OpenCM 485 EXP)

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      100                 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      900                 // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

[... 140 lines cut ...]

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDRESS_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    packetHandler->getTxRxResult(dxl_comm_result);
  }
  else if (dxl_error != 0)
  {
    packetHandler->getRxPacketError(dxl_error);
  }

  // Close port
  portHandler->closePort();

}

void loop() {
  // put your main code here, to run repeatedly:

}

Here is the output: 

[image]