ROBOTIS-GIT/DynamixelSDK #163: [3.5.4][C++][USB2Dynamixel2][MX-106R] Issue : How can I use the 'sync_write' protocol 2.0.

tutamc, 2018-02-16:
- I'm using `DynamixelSDK ver. 3.5.4`;
- I'm using `C++` Language;
- I'm using `USB2Dynamixel2` serial converter;
- I'm using `MX-106R`;
- My issue is how can I use the function "SyncWrite" on my MX-106R (protocol 2.0).

Hey, I work with a humanoid robot that use Dynamixel's actuator. This code was used to control a MX-28 (protocol 1.0), that I called a function "SyncWrite" to move two or more servos at the same time:

----------------------------------------------------CODE------------------------------------------------------------
#define NUMBER_OF_JOINTS 21
#define PARAM_BYTES 7

int param[NUMBER_OF_JOINTS * PARAM_BYTES];
int n = 0;
int joint_num = 0;
for(int id=1; id<=20; id++)
{
  if(MotionStatus::m_CurrentJoints.GetEnable(id) == true)
  {
  param[n++] = id;
  param[n++] = MotionStatus::m_CurrentJoints.GetDGain(id);
  param[n++] = MotionStatus::m_CurrentJoints.GetIGain(id);
  int p_gain = m_voltageAdaptionFactor * MotionStatus::m_CurrentJoints.GetPGain(id);
  if(p_gain <= 0)
      p_gain = 1;
  param[n++] = p_gain;
  param[n++] = 0;
  param[n++] = CM730::GetLowByte(MotionStatus::m_CurrentJoints.GetValue(id));
  param[n++] = CM730::GetHighByte(MotionStatus::m_CurrentJoints.GetValue(id));
  joint_num++;
  }
}
if(joint_num > 0)
m_CM730->SyncWrite(MX28::P_CW_COMPLIANCE_SLOPE, MX28::PARAM_BYTES, joint_num, param);
----------------------------------------------------------------------------------------------------------------

Now I have MX-106R (protocol 2.0) and my issue is how can I rewrite this code to use on my new servos with protocol 2.0.

Thanks,

Lucas.

ROBOTIS-Leon, 2018-02-17:
@tutamc 

Hi,

I don't actually know where the code is come from, but if you want make a code for a humanoid and you don't have any idea how to port SDK to your robot, [ROBOTIS Forum](http://en.robotis.com/service/forum.php) may help you.

From my thoughts. the code you've attached was intended to make 20 MX motors move by handling each of PID gain, goal position. 
To control those which are far from each other's control address, but simultaneously, you should use `indirect address` function to get those items nearby, then `syncwrite` to change the values. You can find `Indirect address` example code in the Dynamixel SDK package.
To use `indirect address` function of MX series, you should update its firmware to the new (it will update automatically when you handle it by RoboPlus 2.0). 
See control table of the MX [here](http://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/)

tutamc, 2018-02-19:
Thanks @ROBOTIS-Leon !!

ROBOTIS-Leon, 2018-02-20:
@tutamc 

You are welcomed

closed 2018-02-20 by ROBOTIS-Leon