ROBOTIS-GIT/DynamixelSDK #164: [USB2Dynamixel][XL430-W250-T] Issue: cannot update Dynamixel on R+ Manager

AdrianFHAachen, 2018-02-27:
We have recently purchased several Dynamixel XL430-W250-T to build a Turtlebot 3. We have installed R+ Manager 2.0 to update the firmware with the desired baudrate and ID. However, we haven't managed to set the Dynamixels.

We are connecting the Dynamixel to a computer via USB2Dynamixel. After we select "Update & Test" on R+ Manager 2.0 the motor is detected correctly. However, afterwards the communication status of RX is shown as "corrupt" and we get a "CRC error". We haven't been able to change the baud rate or the ID. The TX channel blinks and is sometimes shown as green and sometimes as grey.

[image]

When we try to do the firmware recovery the installation always fails. Sometimes it fails at 0%, and sometimes at 99.67%. Afterwards we have the problem that the Dynamixel is not even detected anymore, and the red light of the motor does not turn on when we power it.

[image]

We would greatly appreciate any help on this matter.

ROBOTIS-Leon, 2018-03-05:
@AdrianFHAachen 

Hi,

since here can't handle the issue related with the Robo Plus, I sent this to [robotis forum](http://en.robotis.com/service/forum_view.php?slg=&page_type=&bbs_no=2406716).
you can get the details from there.

ROBOTIS-Leon, 2018-03-09:
@AdrianFHAachen 

Hi, again

you can follow up the status of how your issue is being solved.
That bulletin is currently asking some questions for you.

AdrianFHAachen, 2018-03-09:
Hello,

Thank you for your help. We will ask the question posted soon.

Regards,
Adrian

AminMohammadi, 2018-03-11:
@ROBOTIS-Leon 
I saw Robotis download page, and it seems that R+ manager is only for windows. I only have Linux on my system. Is there any way to change motor ID, operating mode, and everything in Linux? It's inconvenient that I have to go to windows system to just change the motor configuration. If I can do that in Linux, is there a tutorial for beginners on how to do it?

ROBOTIS-Leon, 2018-03-15:
@AminMohammadi 

Hi, 
there are some solutions:

1. If you can use ROS : Download [Dynamixel Workbench](http://wiki.ros.org/dynamixel_workbench) and have a look of how to use it. Dynamixel Workbench is like-a-R+ Manager for Linux, since it supports GUI and a function that makes you be able to change the ID easily as well as you can do in R+ Manager.

2. If you think that you are good at handling DynamixelSDK : mix broadcast ping example and readwrite example. DynamixelSDK has very basic APIs that is likely used in R+ Manager. Thus, you can make your semi-R+ Manager by yourself with DynamixelSDK APIs.

ROBOTIS-Leon, 2018-03-20:
@AdrianFHAachen 

Hi, how are you

since your question has been pretty well settled, this issue will be closed. 
Let me know if you have another question related with this bulletin.

closed 2018-03-20 by ROBOTIS-Leon