ROBOTIS-GIT/DynamixelSDK #165: [3.4.6][C][USB2U2D2][XM430-W350-R] Issue : Error connecting U2D2 via linux

karishmabava, 2018-03-01:
ISSUE TEMPLATE ver. 1.1.1
I have followed all the steps for linux64 ubuntu and getting the following error:-

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!

I have defined the following values  for it:--

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        57600 
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      1024                // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      3074                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

Please help!!!!!!!

ROBOTIS-Leon, 2018-03-05:
@karishmabava 

Hi,

which Dynamixel did you used?
I need more at-the-least details of the circumstance to solve it

ROBOTIS-Leon, 2018-03-05:
@karishmabava

Hi again,
every X series Dynamixel would be set as ID 1, Baudrate 1M

to handle Dynamixel by other than 1M, you should change both Baudrate set in Dynamixel and the Baudrate written in the source code. To change its Baudrate, use Roboplus 2.0 manager. 

See here:
http://en.robotis.com/service/downloadpage.php?ca_id=10

ROBOTIS-Leon, 2018-03-09:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

karishmabava, 2018-04-09:
Hi Leon,

I am using a Linux64 machine and writing in c code. I am able to reset the ids but the baudrate for the dynamixels are giving me problem... can me point to the function I can use?  This is my code
`/ Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132
#define ADDR_ID								 7
#define ADDR_BAUDRATE						 8
//#define ADDR_MAX_VEL                     44
//#define ADDR_GOAL_VEL                 104
//#define GOAL_VELOCITY                 10
#define MAX_ID1									 255
// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          5                   // Dynamixel ID: 1
#define BAUDRATE1                       1000000 
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
// int port_num = 0;                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0                // Dynamixel will rotate between this value

#define DXL_MAXIMUM_POSITION_VALUE      4095                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)

#define ESC_ASCII_VALUE                 0x1b

int getch(){

    struct termios oldt, newt;
    int ch;

    tcgetattr(STDIN_FILENO, &oldt);
    newt = oldt;
    newt.c_lflag &= ~(ICANON | ECHO);

    tcsetattr(STDIN_FILENO, TCSANOW, &newt);
    ch = getchar();
    tcsetattr(STDIN_FILENO, TCSANOW, &oldt);

    return ch;
}

int kbhit(void)
{

    struct termios oldt, newt;
    int ch;
    int oldf;

    tcgetattr(STDIN_FILENO, &oldt);

    newt = oldt;
    newt.c_lflag &= ~(ICANON | ECHO);

    tcsetattr(STDIN_FILENO, TCSANOW, &newt);
    oldf = fcntl(STDIN_FILENO, F_GETFL, 0);
    fcntl(STDIN_FILENO, F_SETFL, oldf | O_NONBLOCK);

    ch = getchar();

    tcsetattr(STDIN_FILENO, TCSANOW, &oldt);
    fcntl(STDIN_FILENO, F_SETFL, oldf);

    if(ch != EOF) {
       ungetc(ch, stdin);
       return 1;
    }

    return 0;
}

 

int main(){
	
	int count =0;
//	int port_num = init_dynamixel();
	
	int port_num = portHandler(DEVICENAME);

   // Initialize PacketHandler Structs
   packetHandler();

   if(!openPort(port_num)){
      printf("Failed to open the port:%d!\n" ,port_num);
      printf("Press any key to terminate...\n");
      getch();
      return 0;
   }

	int dxl_comm_result = COMM_TX_FAIL;

   printf("Now the port  baudrate is : %d\n", getBaudRate(port_num));

	// Set port baudrate
  if (setBaudRate(port_num, BAUDRATE1))
  {
    printf("Succeeded to change the baudrate!\n");
  }
  else
  {
    printf("Failed to change the baudrate!\n");
    printf("Press any key to terminate...\n");
    getch();
    return 0;
  }

	write1ByteTxRx2(port_num, BROADCAST_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE);
	write1ByteTxRx2(port_num, BROADCAST_ID, ADDR_ID, DXL_ID);
	write1ByteTxRx2(port_num, BROADCAST_ID, ADDR_BAUDRATE, BAUDRATE1);

	for(int i=0;i<10000;i++);
	write1ByteTxRx2(port_num, BROADCAST_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_ENABLE);
	
	broadcastPing(port_num, PROTOCOL_VERSION);
  if ((dxl_comm_result = getLastTxRxResult(0, 2.0)) != COMM_SUCCESS)
    printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));

  printf("Detected Dynamixel : \n");
  for (int id = 0; id <= MAX_ID1; id++)
  {
    if (getBroadcastPingResult(port_num, PROTOCOL_VERSION, id))
      printf("[ID:%03d]\n", id);
  }
	

return 0;

}`

closed 2018-04-20 by robotpilot