ROBOTIS-GIT/DynamixelSDK #166: [3.5.4][Arduino]: Support other Arduino boards?

KurtE, 2018-03-01:
- I'm using `DynamixelSDK ver. 3.5.4`
- I'm using `Arduino` Language
- I'm using `OpenCM904` serial converter
- I'm using `XM430-W250-T` and AX-12 and AX-18

I am wondering if your are interested in supporting additional boards that are supported by the Arduino IDE.

In my particular cases, I am interested in ARM based Teensy boards by PJRC (https://www.pjrc.com/teensy/) such as Teensy 3.2, 3.5, 3.6, LC.

In addition potentially support some AVR based boards, such as the Arbotix-M (http://www.trossenrobotics.com/p/arbotix-robot-controller.aspx) by Trossen Robotics.   

Teensy:
I have a version of the DynamixelSDK that it at least partially working on a T3.2 and a 3.6 board.  A lot of this was modeled on my own Dynamixel support library (https://github.com/KurtE/BioloidSerial) which was based on earlier Bioloid library by Trossen Robotics/Interbotix.  

There are two different ways I support the half duplex on the Teensy boards.  Either by using the half duplex support built into their Uarts, or by using an external TTL buffer(s), which interfaces in like an RS485 support chip with a signal direction support pin... 

To support this I added a method: PortHandlerArduino::setTXEnable(int pin); 

Still testing: Current WIP is up at: https://github.com/KurtE/DynamixelSDK/tree/Support_Teensy

AVR: 
Several issues:  Not sure how much it is worth to do?

#include \<vector\> and \<map\> - Arduino IDE does not install STL libraries.  So can either #fdef out all uses  or use the Arduino library manager and install: ArduinoSTL.

mismatch in return value from Protocol2PacketHandler updateCRC.  uint16_t versus unsigned short.

Lots of signed vs unsigned compare: for (int i=0; i < sizeof(XXX); i++) warnings:

Problems with Baud rate apis: like: 

    int PortHandlerArduino::checkBaudrateAvailable(int baudrate)

Problem is that on AVR int will be 16 bits, so: 1000000 will not fit  So the API should probably be changed to either: uint32_t or int32_t.

Likewise constants like: 

    return 1000000 ;

Will give at least compile warning and maybe not work properly, in these cases should be something like:

    return 1000000L;

So again wondering if you would be interested in these levels of code changes?

OpusK, 2018-03-06:
Hello, @KurtE 

There are no plans yet for additional board support.

Regarding AVRs, porting is very difficult due to insufficient memory and is possible on PC and CM9.04 level or higher.
This is because the Dynamixel-SDK was originally designed for PCs.

In fact, recently we are building a Dynamixel Shield library to support Arduino boards.
Of course, as mentioned earlier, due to memory issues, it is a different library from the SDK.
It is still unofficial, so please refer to [this link](https://github.com/ROBOTIS-GIT/DynamixelShield/tree/master) if you want. ([wiki](https://github.com/ROBOTIS-GIT/DynamixelShield/wiki))

KurtE, 2018-03-06:
Thanks, 

 

Would you be interested into merging in code for other ARM based boards like the Teensy?

 

Again I have a version that is working.  Probably would take another pass to clean it up and remove some of the AVR stuff out.

 

Yes the AVR is limited. The Atmega644P that the Arbotix-M board uses has 64K storage and 4K ram… I did have a version of your library that built for ARM, but it used more RAM.  Some of which was due to my sloppy usage of memory in test app, so could probably make it work.

 

What I am trying to decide in my own code is to stick with my own libraries, or to try to converge on using your library… I do have a version of my Hexapod code base that works on the openCM9.04 with the Trossen Robotics PhantomX robot… Would be nice to have one set of code that works with all of the different controllers… 

 

Will take a look at your AVR code.  Hopefully it is simply a stripped down version of your main library, with maybe some of the memory intensive things removed, or errors out. 

Example:  removes all of the Protocol 2 things that depends on vectors and maps… The rest should be pretty simple. 

 

Thanks again

Kurt

OpusK, 2018-03-07:
@KurtE, thank you for your passion and efforts.

We have discussed similar issue with yours in the past. However, due to company policy, the SDK does not support embedded systems. (Except OpenCR and OpenCM9.04 that have already been created)

This repository and OpenCR and OpenCM 9.04 are managed separately.
In other words, we can not merge or manage other code because we do not manage the integrated SDK for embedded by company policy.
I wish I could merge your well-made code, but unfortunately I can not.

If we are going to support and manage the embedded SDK separately, we will take your comments into consideration.

Thanks again for your passion and efforts.

NicoXcc, 2018-03-10:
Robotis team, 
I stand second to @KartE as Teensy 3.6 is a really nice board. it would be really nice if we can use Teensy with latest Dxls like X Series with oldest AX series. I know that there are some company policies that we need to adhere to but I certainly feel I would mention I found a near similar effort done by another community member.
https://github.com/ROBOTIS-GIT/OpenCR/issues/26

I am not sure if I am going off-topic :)
NicoX

ROBOTIS-Leon, 2018-03-15:
@robotpilot @chcbaram 

How do you think of this?

KurtE, 2018-03-15:
Thanks guys,

Let me know if I can help in any way.   

For what it is worth, as I mentioned, I have added some support for Protocol 2, to my own version of the earlier Bioloid library by Trossen Robotics (Interbotix).   Currently the protocol 2 stuff is in a working branch: https://github.com/KurtE/BioloidSerial/tree/WIP-T2

As I also mentioned earlier I have had some success making a version of the DynamixelSDK that is working for the Teensy.  So far i have tried on T3.2 and 3.6.   If it looks promising there are a few additional setup things I would probably want to do, including: On the Teensy 3.5 and 3.6 support all 6 Serial ports, although again still more likely to use Serial1 and Serial2 as hardware wise they have more buffering and can go to higher speeds.    

I would also add in better support for alternate pins.  What I mean by this is that on a Teensy, by default certain functionality defaults to specific pins.  Example for Serial1 the RX and TX pins are on pins 0 and 1, but the hardware may allow you to use different pins for these.  Example I could set the TX to pin 26... 
The issue in my setup code, is if I am configuring to use only the TX pin for the half duplex, there are things I typically do to configure the pin, like enable the pullup resistor.  Currently this code assume default TX pin...

And there there are a few other code changes I might make, like in the Teensy case hold on to a pointer to the hardwareSerial class instead of the generic Stream class, so you dont have to figure out which serial class you are using to be able to call begin and end which are not part of the Stream class.

Again let me know if I can help.  Currently I am a bit distracted from this as I am now assembling a Turtlebot3 Waffle Pi.  I also purchased the Intel Real-sense SDK kit with the R200 camera and UP board), which once I get things working well with the RPI will probably swap in the UP...

NicoXcc, 2018-03-15:
I would love to see Robotis takes interests on Teensy. Appreciate your hard work KartE.
As discussed in the other forum (Trossen Robotics) I am also interested to see where these hard efforts fly up. Let me know if I can help you in any, as I do have a Teensy 3.6 setup with my DSO on desk.

OpusK, 2018-03-21:
@ROBOTIS-Leon.

As a result of the in-team meeting, we concluded that support for this issue is still difficult due to company policy.

ROBOTIS-Leon, 2018-03-21:
@OpusK 

Okay,

@KurtE @NicoXcc 

I'm sorry to say that it might not be done this time.
But surely we will discuss further internally.

KurtE, 2018-03-22:
@ROBOTIS-Leon  

Thanks, 

Looks like for now I will maintain my own version.  Probably will split off and create new github project probably with only the Arduino needed stuff.

NicoXcc, 2018-03-22:
@KartE
I feel you may want to upload circuit diagrams in your teensy repo for so that others can follow and share there experience with Dynamixel actuators.

KurtE, 2018-03-22:
@NicoXcc - Warning I am not a EE type but instead a retired Software guy and I do all of this for my own playing around.  So some things work and somethings don't  

But I do have several of the designs up in the github project: https://github.com/KurtE/Teensy3.1-Breakout-Boards

Note: this is a Hodge Podge of stuff.  Many of the boards I made were never full tested... But I do have fun. 

In most cases I have the Diptrace design files up there, plus a zip file with the Gerber and Drill files.  These zip files are what I upload to do fabrications at places like OSHPark (most often), sometimes Seeedstudio and a few times pcbs.io 

Again all of these designs are use at your own risk, no guarantees that they work...

NicoXcc, 2018-03-22:
Thanks @KurtE ... dhuh first time I took the correct name, so no need to edit post :)
No worries, it was also mentioned in the readme of your repo. Currently I am tuning faster communication with Teensy 3.6 for my bot. Will share once finished ... again I was doing on my risk for sure.

ROBOTIS-Leon, 2018-03-30:
@KurtE @NicoXcc 

Hi,

I will put "won't fix" label on this issue but will leave "may be this help you" label anyway here.

BTW, I don't want to let your splendid works remain obscure. 
https://community.robotsource.org/ is a page that many users & developers like you are having the interests in the field of your work. Please share your ideas, make your idea much better.

Cheers

NicoXcc, 2018-03-31:
@ROBOTIS-Leon
Thanks for your suggestions.
I put some of my work here with a suggestion to upgrade AX series https://community.robotsource.org/t/can-we-rebirth-ax-12-with-x-series-specs/1376/3

ROBOTIS-Leon, 2018-04-09:
@NicoXcc 

Great, 
I will close this issue, but I will leave the label `enhancement` to let someone also are interested on this issue to track your further debates. See you there

closed 2018-04-09 by ROBOTIS-Leon