ROBOTIS-GIT/DynamixelSDK #170: PInvokeStackImbalance errors in Visual Studio with SDK 3.5.4

Miedlar, 2018-03-08:
I'm not sure3 this is the proper place to put this, but if you run into this, please be aware that this is a Microsoft issue.  You can turn off this "feature" but going into the "Managed Debuggunbg Assistants" (hit ctrl-alt-E to get to it) and uncheck that value.  I hope this helps!

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Leon, 2018-03-09:
@Miedlar 

Hi, 
thank you for the advice

I wondered if you were trying to do build the SDK C#, and saw PInvokeStackImbalance exception caused by CallingConvention problem.

Miedlar, 2018-03-09:
Leon,

I wasn't trying to build the SDK itself, I was just using the C dll via
Import DLL.  I was basing my code on the ping and read-write samples
included with the SDK.   But yes, I think the Calling Convention was in
play here.

I hope that helps.  If you need more info, please let me know.

BTW, according to the research I did, this was something that crept in
between Visual Studio 2008 and 2010.

M**2

ROBOTIS-Leon, 2018-03-09:
@Miedlar 

Understood. I will leave this on closed issue, since someone who wants to get their answers used to search information either from closed issue.

do you agree?

Miedlar, 2018-03-09:
Leon,

Absolutely, I only put in there at Brandon's request, I wasn't sure it was
the proper place to put it.  But I think we wanted me to share what I'd
found with y'all.  I hope it's of use to you.

M**2

Miedlar, 2018-03-09:
Leon,

If it helps, take a look at:

https://stackoverflow.com/questions/3506796/pinvokestackimbalance-how-can-i-fix-this-or-turn-it-off

That tells more about what's up and what you (and I do mean you, as in you
at Robotis) can do about it.

M**2

ROBOTIS-Leon, 2018-03-15:
@Miedlar 

Thank you for great ideas from you
It seems like the SDK DLL C ver, C# porting should be modified in next release

Miedlar, 2018-03-15:
Leon,

I'm glad it helped.

M**2

ROBOTIS-Leon, 2018-05-15:
@Miedlar 

Hi, again,

since the visual studio 2008 is no longer available via official download center in microsoft page, 
we are thinking no many related issues will be made afterwards. 
but anyway the users who will see invoke problem, will see this issue, though this is closed.

Thank you for the advice, again, and let me close this issue.

closed 2018-05-15 by ROBOTIS-Leon

Miedlar, 2018-05-15:
Leon,

I am uncertain what Vis Studio 2008 has to do with this, I use 2015 myself
and the issue stil exists in there.

M**2