ROBOTIS-GIT/DynamixelSDK #171: [3.5.4][C++][USB2Dynamixel][XM430-W210-T] Issue : Reverse rotation

AminMohammadi, 2018-03-11:
@ROBOTIS-Leon 
Hello
I want to rotate my dynamixel in reverse direction, but it gives me this error:

[RxPacketError] Writing or Reading is not available to target address!

On the other hand, documentation says that "Drive Mode" is "RW", so I should be able to change direction in my code. I know that I can connect it to R+ Manager to do it, but sometimes I want to change direction during the code. Here is emanual "http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#drive-mode". The code is in the .txt file

[velocity.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/1800976/velocity.txt)

My second question is is there any way to change motor configuration in Linux? If yes, is there a tutorial for that? I only have Linux on my system, and R+ Manager works only in Windows. I saw that in tutorial videos and download page that R+ Manager works only on windows.

Thanks(I mentioned my hardware info in the headline of question)

AminMohammadi, 2018-03-15:
solved. I just need to send a negative value

ROBOTIS-Leon, 2018-03-15:
@AminMohammadi 

Hi, 
sorry for the late reply,

good to see that you've found your solution.

closed 2018-03-15 by ROBOTIS-Leon