ROBOTIS-GIT/DynamixelSDK #174: [python][MX28] -  Bulk read multiple entities from an array of motor

vikashplus, 2018-03-23:
I'm trying to read multiple entities (4 entities: position, velocities, load, moving) from nine MX28 motors. What's the most efficient way to get all the values?

1) I tried to use groupBulkReadAddParam to add all the desired parameters (9*4=36 params) to a groupBulkRead but that didn't work.

2) I, however, got some success in making one group per entity (4 groups: position, velocities, load, moving). But then I need to issue 4 groupBulkRead instead of just one which is inefficient.

Please advise what's the best way to achieve this. This seems like a general use case, please consider adding an example script.

Thanks

ROBOTIS-Leon, 2018-03-27:
@vikashplus 

Hi,
sorry for the late reply

I recommend you use SyncRead function with Indirect Address (This function must be updated if your MX has recently upgraded its Firmware. I still don't know if you are using it on Protocol 2.0 or not, but if it is on Protocol 2.0, you will be pleased since it will get another various functions, too). 

SyncRead will run the multiple entities of more than one Dynamixel with packet shorter than BulkRead ( = faster communication). SyncRead, exactly saying, can't read the entities which are not nearby, so if you copy the entities to the Indirect Address / Indirect Data to make them be nearby, you can read all of them by SyncRead.

See indirect address / sync read write example code.

vikashplus, 2018-03-27:
I'm on Protocol 1.0. 

My understanding is that indirect addressing isn't available for Protocol 1.0. What will you advise for protocol 1?

ROBOTIS-Leon, 2018-03-27:
@vikashplus 

Hi, again.

Currently, BulkRead will allow you add only one groupbulkread per Dynamixel (it watches whether you've already added groupbulkread to the same ID of Dynamixel), so you can't add for each entities. 

Since protocol1.0 doesn't have syncread, but MX 1.0 has indirect address (if you updated new firmware), add one groupbulkread that will read all entities in the indirect address / data. Of course you should copy all the data of targeted entities into the indirect address / data. This is the same use of what I recommended as above.

vikashplus, 2018-03-27:
I have never done indirect accessing. Can you point me to an example script?

ROBOTIS-Leon, 2018-03-28:
@vikashplus 

[this is what you requested](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/example/protocol2.0/indirect_address/indirect_address.cpp)

vikashplus, 2018-03-28:
"Since protocol1.0 doesn't have syncread, but MX 1.0 has indirect address" --- 

I'm a little confused. This example script is for protocol 2.0. Can I use it with protocol 1.0 as well?

ROBOTIS-Leon, 2018-03-29:
@vikashplus 

Yes, but that uses syncread function, but protocol 1.0 doesn't have it, so you should replace in to bulkreads

closed 2018-04-09 by ROBOTIS-Leon