ROBOTIS-GIT/DynamixelSDK #175: [3.5.4][C++][USB2Dynamixel][XM540-W270-R] Issue: RxPacketError - Data is out of range

baertscc, 2018-03-26:
Hi there, I've got the exact same problem as the guys [here  ](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/148)

Which DYNAMIXEL is it? [XM540-W270-R]
Which CONTROLLER is it? [PC via USB2Dynamixel]
Which VERSION of DynamixelSDK is it? [3.5.4 - C++]

> Succeeded to open the port!
> Succeeded to change the baudrate!
> [RxPacketError] The data value is out of range! Press any key to continue! (or press ESC to quit!
> [RxPacketError] The data value is out of range!
> [ID:001] GoalPos:000 PresPos:-1945990398
> [ID:001] GoalPos:000 PresPos:-1945990398
> [ID:001] GoalPos:000 PresPos:-1945990398

When using the R+ Manager, the Dynamixel works fine and all Rx  Packages contain the correct data. When using the SDK, it always leads to this bug. The error is consistently reproducible under Windows (Win 10) and Linux (Ubuntu 16.04).

ROBOTIS-Leon, 2018-03-27:
@baertscc 

Hi,

can you paste the code you've tried with?

baertscc, 2018-03-27:
The problem arose when trying to execute the [read-write example](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/example/protocol2.0/read_write/read_write.cpp) script

ROBOTIS-Leon, 2018-03-28:
@baertscc 

Hi, again,

you should modify the source code yourself:
from [this line](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/200dcd2cc5034257ccad1f0132f1dae9b6b398a9/c%2B%2B/example/protocol2.0/read_write/read_write.cpp#L41)

I will attach some codes. Replace the following to each appropriate place .These will run the X series Dynamixel. 

``` cpp
// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132
```

if you changed anything in the Dynamixel after it was given, change these

``` cpp
// Default setting
#define DXL_ID                              1                   // Dynamixel ID: 1
#define BAUDRATE                        57600
```

if you are using Windows or Mac, these should be edited.

``` cpp
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

```

Maximum and Minimum position of the X series is 0 ~ 4096, so you should modify these.

``` cpp
#define DXL_MINIMUM_POSITION_VALUE      100             // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4000              // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
```

give me feedbacks after doing these.

baertscc, 2018-03-28:
Thank you very much!
I forgot to set the Control table address to the correct values. Now everything works perfectly!

Thanks again for the competent and responsive support!

ROBOTIS-Leon, 2018-03-28:
@baertscc 

Hi,
so pleased to hear that.

tell me when you have another questions.

BR

closed 2018-03-28 by ROBOTIS-Leon

hamidnrn, 2018-09-05:
I have the same problem with matlb , I'm using the XH430-W210 ! it's not moving at all and keeps giving me this error !!
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or input e to quit!
[RxPacketError] The data value is out of range!
[ID:001] GoalPos:100  PresPos:-1945990398
[ID:001] GoalPos:100  PresPos:-1945990398