ROBOTIS-GIT/DynamixelSDK #179: [3.5.4][MATLAB][USB2Dynamixel2][AX-18A] Issue : Where to access all registers

taylormr, 2018-04-06:
I'm using AX-18A motors in MATLAB with U2D2 and have found functions such as writeAngle, setSpeed, setMode, etc. in the "MyDynamixel" file. How can I access the other registers in MATLAB to just read things like present position, present speed, etc? I'm guessing it has something to do with accessing the control table?

Thanks!

ROBOTIS-Leon, 2018-04-07:
@taylormr 

Hi,

please refer to the read_write.m example source code; it writes goal position, reads present position on every sequence. 

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/matlab/protocol1.0/read_write.m

To read present speed, see 

http://emanual.robotis.com/docs/en/dxl/ax/ax-18a/#control-table-of-eeprom-area

and change ADDR on the source code to 32, which is the control table address of moving speed

ROBOTIS-Leon, 2018-04-13:
@taylormr

Which version of MATLAB do you use?

taylormr, 2018-04-13:
R2017a

ROBOTIS-Leon, 2018-04-16:
@taylormr 

Hi, again

did you tried the source in develop branch?

taylormr, 2018-04-16:
No I used the master branch. Should I be in the develop branch?

I'm getting the following error now after updating to R2018a.

C:/ProgramData/MATLAB/SupportPackages/R2018a/3P.instrset/mingw_w64.instrset/x86_64-w64-mingw32/include/stdint.h:35:21: error:
conflicting types for 'int8_t'
 typedef signed char int8_t;

taylormr, 2018-04-16:
I tried the develop branch and it solved my issue! Thank you so much, I really appreciate it.

ROBOTIS-Leon, 2018-04-17:
@taylormr

Hi, and glad to see it works.
Cool.

closed 2018-04-17 by ROBOTIS-Leon