ROBOTIS-GIT/DynamixelSDK #183: Unable to read Present speed of xl-320

Lazy12316, 2018-04-30:
Hi, 

- I'm using opencm9.04c to control xl-320 servo motor.
- I'm unable to read the present speed and torque of the xl-320 in joint mode.
- I'm able to get the feedback from xl-320 when i set the moving speed manually to address 32 of xl-320 in wheel mode.
- when i run the motor in joint mode and try to read the present speed, i get the present speed as 0.
- The torque value when read by the motor gives a value of 1023 even when i change the speed of the motor manually(i.e, by giving values to the motor address). 

Please let me know, how to rectify the issue as soon as possible.

Thanks.

ROBOTIS-Leon, 2018-05-02:
@Lazy12316 

Hi,

since here can't handle the issue you've commented, I will send this to [robotis forum](http://en.robotis.com/service/forum_view.php?slg=&page_type=&bbs_no=2406786).
you can follow up the status of how your issue is being solved.

Lazy12316, 2018-05-02:
@ROBOTIS-Leon 
Thank You.

ROBOTIS-Leon, 2018-05-04:
@Lazy12316 

You're welcomed

closed 2018-05-04 by ROBOTIS-Leon