ROBOTIS-GIT/DynamixelSDK #184: Error in [TxRxResult] There is no status packet! in MAC OSX High Sierra

martab94, 2018-05-03:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using the latest `DynamixelSDK develop'

- I'm using `MATLAB 2018a` Language

- I'm using `USB2Dynamixel2` serial converter

- I'm using `AX-12A`

- and I'm having an `issue` like what

  - '[TxRxResult] There is no status packet!
[ID:032] GoalPos:400  PresPos:000
[TxRxResult] There is no status packet!
[ID:032] GoalPos:400  PresPos:000
[TxRxResult] There is no status packet!
[ID:032] GoalPos:400  PresPos:000'

Hi, as said I want to connect a DYNAMIXEL AX12A motor with Matlab 2018a on my MAC OSX, but I got the error  '[TxRxResult] There is no status packet!' while testing read_write.m example. I have already changed the baudrate, the ID, the port number and the goal position according to the e-manual. 

I attached my code below.

Thanks in advance.

clc;
clear all;

lib_name = '';

if strcmp(computer, 'PCWIN')
  lib_name = 'dxl_x86_c';
elseif strcmp(computer, 'PCWIN64')
  lib_name = 'dxl_x64_c';
elseif strcmp(computer, 'GLNX86')
  lib_name = 'libdxl_x86_c';
elseif strcmp(computer, 'GLNXA64')
  lib_name = 'libdxl_x64_c';
elseif strcmp(computer, 'MACI64')
  lib_name = 'libdxl_mac_c';
end

% Load Libraries
if ~libisloaded(lib_name)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h', 'addheader', 'packet_handler.h');
end

% Control table address
ADDR_AX_TORQUE_ENABLE       = 24;           % Control table address is different in Dynamixel model
ADDR_AX_GOAL_POSITION       = 30;
ADDR_AX_PRESENT_POSITION    = 36;

% Protocol version
PROTOCOL_VERSION            = 1.0;          % See which protocol version is used in the Dynamixel

% Default setting
DXL_ID                      = 32;            % Dynamixel ID: 1
BAUDRATE                    = 57600; 
DEVICENAME                  = '/dev/tty.usbserial-A4008aCg';       % Check which port is being used on your controller
                                            % ex) Windows: 'COM1'   Linux: '/dev/ttyUSB0' Mac: '/dev/tty.usbserial-*'

TORQUE_ENABLE               = 1;            % Value for enabling the torque
TORQUE_DISABLE              = 0;            % Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 400;          % Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 700;         % and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
DXL_MOVING_STATUS_THRESHOLD = 10;           % Dynamixel moving status threshold

ESC_CHARACTER               = 'e';          % Key for escaping loop

COMM_SUCCESS                = 0;            % Communication Success result value
COMM_TX_FAIL                = -1001;        % Communication Tx Failed

% Initialize PortHandler Structs
% Set the port path
% Get methods and members of PortHandlerLinux or PortHandlerWindows
port_num = portHandler(DEVICENAME);

% Initialize PacketHandler Structs
packetHandler();

index = 1;
dxl_comm_result = COMM_TX_FAIL;             % Communication result
dxl_goal_position = [DXL_MINIMUM_POSITION_VALUE DXL_MAXIMUM_POSITION_VALUE];         % Goal position

dxl_error = 0;                              % Dynamixel error
dxl_present_position = 0;                   % Present position

% Open port
if (openPort(port_num))
    fprintf('Succeeded to open the port!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to open the port!\n');
    input('Press any key to terminate...\n');
    return;
end

% Set port baudrate
if (setBaudRate(port_num, BAUDRATE))
    fprintf('Succeeded to change the baudrate!\n');
else
    unloadlibrary(lib_name);
    fprintf('Failed to change the baudrate!\n');
    input('Press any key to terminate...\n');
    return;
end

% Enable Dynamixel Torque
write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_AX_TORQUE_ENABLE, TORQUE_ENABLE);
dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
if dxl_comm_result ~= COMM_SUCCESS
    fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
elseif dxl_error ~= 0
    fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
else
    fprintf('Dynamixel has been successfully connected \n');
end

while 1
    if input('Press any key to continue! (or input e to quit!)\n', 's') == ESC_CHARACTER
        break;
    end

    % Write goal position
    write2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_AX_GOAL_POSITION, dxl_goal_position(index));
    dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
    dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
    if dxl_comm_result ~= COMM_SUCCESS
        fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
    elseif dxl_error ~= 0
        fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
    end

    while 1
        % Read present position
        dxl_present_position = read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_AX_PRESENT_POSITION);
        dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
        dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
        if dxl_comm_result ~= COMM_SUCCESS
            fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
        elseif dxl_error ~= 0
            fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
        end

        fprintf('[ID:%03d] GoalPos:%03d  PresPos:%03d\n', DXL_ID, dxl_goal_position(index), dxl_present_position);

        if ~(abs(dxl_goal_position(index) - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD)
            break;
        end
    end

    % Change goal position
    if index == 1
        index = 2;
    else
        index = 1;
    end
end

% Disable Dynamixel Torque
write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_AX_TORQUE_ENABLE, TORQUE_DISABLE);
dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION);
dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
if dxl_comm_result ~= COMM_SUCCESS
    fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
elseif dxl_error ~= 0
    fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
end

% Close port
closePort(port_num);

% Unload Library
unloadlibrary(lib_name);

close all;
clear all;

ROBOTIS-Leon, 2018-05-04:
@martab94 

Hi,

1. Could you recheck if the Dynamixel settings are the same as you set in the code?

2. Did you put the power in the Dynamixel over 12V?

3. Could you `ping.m` the Dynamixel?

4. Does it move during the repetition?

martab94, 2018-05-04:
Hi, 

1. I checked and they are exactly the same. I also tried the code on Windows 10 and Matlab 18a and it works perfectly, the motor moves, but not on the Mac that I should use. 

2. Yes it is set at 12V.

3. If I ping.m the Dynamixel I get:

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!

4. So it does not move

I really hope there is a solution!

ROBOTIS-Leon, 2018-05-04:
@martab94 

Okay, then the problem must be related to the port_handler_mac, since your mac will use same codes that is used in the windows as well, except the port handler.

1. Check if read_write.c works in your mac

2. if it works, then perhaps Matlab2018 @ mac or Matlab2018 itself was changed again! into weird way XD

give me feedbacks, please

robotpilot, 2018-05-11:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-05-11 by robotpilot

hamidnrn, 2018-08-28:
Hi i'm using matlab to run DynamixelXH430-W210 , i used the protocol 1 and I faced this error ,
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet

it would be highly appreciated if you could help me resolve this issue

prefpkg21, 2019-01-24:
I increased the value of Return Delay Time  and no longer get status packet errors

amit-420, 2021-09-03:
Doing Firmware recovery using dynamixel wizard worked for me, I was getting the same error while using dynamixel workbench

manueluchiha, 2022-06-07:
Hi, as said I want to connect a DYNAMIXEL AX12A motor with Matlab 2021a on my Lenovo windows, but I got the error '[TxRxResult] There is no status packet!' while testing read_write.m example. I have already changed the baudrate, the ID, the port number and the goal position according to the e-manual.