ROBOTIS-GIT/DynamixelSDK #187: Add python to dynamixel_sdk ROS package

roncapat, 2018-05-15:
Hi @ROBOTIS-Leon, I recently started to learn and work a bit more with ROS and its package system. Now that a Python port of the SDK is starting to take a good shape, I suggest to think about how to make it available as ROS package. 

Currently, only the /c++ subfolder in this repo has a CMakeLists.txt and a package.xml, with ROS package name _dynamixel_sdk_.

Few requirements IMHO:
* the actual package name should not be changed
* the python version should not be distributed as another package with alternate name

Any thoughts? The simple solution would be moving the python folder in the existing package, but this would violate the separation of languages between folders in this repo...

ROBOTIS-Leon, 2018-05-16:
@roncapat 

This is not perfectly decided thing, but perhaps the folder (the repository) will be divided into the non-ROS and for-ROS.

in non-ROS, the python package will be as it is, and aims to pypi release. 
in for-ROS, the python package will get into src folder, so that the folder structure can be really for-ROS.
of course, setup.py will be modified to ROS setup.py

robotpilot, 2018-07-16:
I am thinking of providing a separate ROS package for Python.

robotpilot, 2018-07-17:
I created a new folder called ros for both the C ++ and Python versions of the ROS package and put both languages here (v3.6.2). https://github.com/ROBOTIS-GIT/DynamixelSDK/pull/211

ROS users can now use both the C ++ version of the library and the Python version of the module. Our team will make use of it as an example in [dynamixel_workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench).

The new version of dynamixel_sdk's ROS package is currently awaiting registration.
ROS Indigo: https://github.com/ros/rosdistro/pull/18514
ROS Kinetic: https://github.com/ros/rosdistro/pull/18515
ROS Melodic: https://github.com/ros/rosdistro/pull/18516

robotpilot, 2018-07-17:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue.
Thank you for your consideration and supports. :)

closed 2018-07-17 by robotpilot