ROBOTIS-GIT/DynamixelSDK #189: [3.5.4][Python][USB2Dynamixel2][MX-28] Issue : Bulk read function does not work

pipsqueaker, 2018-05-16:
Hello,

I am attempting to use the BulkRead functionality to simultaneously read the position of many MX-28 motors at once (I realize SyncRead is designed for this, but I intend to BulkRead other values at the same time as well)

However, I am having issues getting BulkRead to work at all. In the following snippet of code, I am attempting to read just the current position of a single motor using BulkRead

```
import ctypes

import dynamixel_functions as dynamixel                     # Uses Dynamixel SDK library

# Control table address
ADDR_MX_TORQUE_ENABLE       = 24
ADDR_MX_PRESENT_POSITION    = 36

# Data Byte Length
LEN_MX_PRESENT_POSITION     = 2

# Protocol version
PROTOCOL_VERSION            = 1

# Default setting
DXL1_ID                     = 18                             # Dynamixel ID: 1
BAUDRATE                    = 1000000
DEVICENAME                  = "/dev/ttyUSB0".encode('utf-8')

TORQUE_ENABLE               = 1                             # Value for enabling the torque
TORQUE_DISABLE              = 0                             # Value for disabling the torque

COMM_SUCCESS                = 0                             # Communication Success result value
COMM_TX_FAIL                = -1001                         # Communication Tx Failed

# Initialize PortHandler Structs
# Set the port path
port_num = dynamixel.portHandler(DEVICENAME)

# Initialize PacketHandler Structs
dynamixel.packetHandler()

dxl_comm_result = COMM_TX_FAIL                              # Communication result
dxl_addparam_result = 0                                     # AddParam result
dxl_getdata_result = 0                                      # GetParam result

dxl_error = 0                                               # Dynamixel error

# Open port
if dynamixel.openPort(port_num):
    print("Succeeded to open the port!")
else:
    print("Failed to open the port, press enter to terminate")
    input()
    quit()

# Set port baudrate
if dynamixel.setBaudRate(port_num, BAUDRATE):
    print("Succeeded to change the baudrate!")
else:
    print("Failed to change the baudrate, press enter to terminate")
    input()
    quit()

# Disable Dynamixel#1 Torque
dynamixel.write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL1_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE)
dxl_comm_result = dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION)
dxl_error = dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION)
if dxl_comm_result != COMM_SUCCESS:
    print(dynamixel.getTxRxResult(PROTOCOL_VERSION, dxl_comm_result))
elif dxl_error != 0:
    print(dynamixel.getRxPacketError(PROTOCOL_VERSION, dxl_error))
else:
    print("Dynamixel#1 has been successfully connected")

group_num = dynamixel.groupBulkRead(port_num, PROTOCOL_VERSION)

# Add parameter storage for Dynamixel#1 present position value
dxl_addparam_result = ctypes.c_ubyte(dynamixel.groupBulkReadAddParam(group_num, DXL1_ID, ADDR_MX_PRESENT_POSITION, LEN_MX_PRESENT_POSITION)).value
if dxl_addparam_result != 1:
    print("[ID:%03d] groupBulkRead addparam failed" % (DXL1_ID))
    quit()

while 1:

    print(dynamixel.read2ByteTxRx(port_num, PROTOCOL_VERSION,
                                  DXL1_ID, ADDR_MX_PRESENT_POSITION))

    # Bulkread present position and moving status
    dynamixel.groupBulkReadTxRxPacket(group_num)
    dxl_comm_result = dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION)
    if dxl_comm_result != COMM_SUCCESS:
        print(dynamixel.getTxRxResult(PROTOCOL_VERSION, dxl_comm_result))
        continue

    # Check if groupbulkread data of Dynamixel#1 is available
    dxl_getdata_result = ctypes.c_ubyte(
        dynamixel.groupBulkReadIsAvailable(group_num,DXL1_ID,
                                           ADDR_MX_PRESENT_POSITION,
                                           LEN_MX_PRESENT_POSITION)).value
    if dxl_getdata_result != 1:
        print("[ID:%03d] groupBulkRead getdata failed" % (DXL1_ID))
        continue

    # Get Dynamixel#1 present position value
    dxl1_present_position = dynamixel.groupBulkReadGetData(group_num, DXL1_ID, ADDR_MX_PRESENT_POSITION, LEN_MX_PRESENT_POSITION)

# Close port
dynamixel.closePort(port_num)
```

Please note that this code is a stripped down version of the example code provided [here](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/python_bulk_read_protocol_1_0/#python-bulk-read-protocol-10), with the exception of the few lines where I read the motor's position using the regular `read2ByteTxRx` function; the example code doesn't work for me either.

When I run this function, I get the following output

```
$ python bulk_thing.py
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel#1 has been successfully connected 
256                                                                       # Produced by single read
[TxRxResult] There is no status packet!            # Produced by attempting to send bulk read packet
256
[TxRxResult] There is no status packet!
256
[TxRxResult] There is no status packet!
256
[TxRxResult] There is no status packet!
256

etc...
```
I can read the correct positions correctly (and can see them change as I move the motor), yet every single attempt I make to send the bulkread packet fails.

ROBOTIS-Leon, 2018-05-17:
@pipsqueaker 

Hi, 

DynamixelSDK Python was completely changed into python originated since version 3.6.0, which is released yesterday.
Try it and give us feedback.

Thank you.

pipsqueaker, 2018-05-17:
Unfortunately the issue persists. I have refactored my code to version `3.6.0` as follows

```
from dynamixel_sdk import *

import itertools

class MultiReader():

    def __init__(self, port_handler, packet_handler, motor_ids, attrs):

        self.port_handler = port_handler
        self.packet_handler = packet_handler
        self.motor_ids = motor_ids
        self.attrs = attrs

        self.param_list = list(itertools.product(self.motor_ids, self.attrs))

        self.bulk_packet = None
        self.construct_packet()

    def construct_packet(self):
        self.bulk_packet = GroupBulkRead(self.port_handler, self.packet_handler)
        for (motor_id, attr) in self.param_list:
            if not self.bulk_packet.addParam(motor_id, *attr):
                raise RuntimeError("Couldn't add parameter storage for "
                                   + str((motor_id, attr)))

    def read(self):

        comm_result = self.bulk_packet.txRxPacket()
        if comm_result != COMM_SUCCESS:
            raise RuntimeError(self.packet_handler.getTxRxResult(comm_result))
        results = [None] * len(self.param_list)

        for index, (motor_id, attr) in enumerate(self.param_list):

            if not self.bulk_packet.isAvailable(motor_id, *attr):
                raise RuntimeError("Data unavailable for " + str((motor_id, attr)))

            results[index] = self.bulk_packet.getData(motor_id, *attr)

        return results

if __name__ == "__main__":

    import motors.p1mx28 as mx28

    PROTOCOL_VERSION = 1
    BAUD = 1000000
    dxl_ids = [12, 18]

    read_attrs = [(mx28.ADDR_PRESENT_POSITION, mx28.LEN_PRESENT_POSITION)]

    port_handler = PortHandler("/dev/ttyUSB0")
    if not port_handler.openPort():
        raise RuntimeError("Couldn't open port")
    if not port_handler.setBaudRate(1000000):
        raise RuntimeError("Couldn't change baud rate")

    packet_handler = PacketHandler(PROTOCOL_VERSION)

    reader = MultiReader(port_handler, packet_handler, dxl_ids, read_attrs)
    print(reader.read())

```

(I added some extra stuff in between, but this code with the parameters set as they are
is functionally equivalent to the code in the issue)

Unfortunately, trying to run it produces the following result

```    
Traceback (most recent call last):
  File "dabs.py", line 129, in <module>
    print(reader.read())
  File "dabs.py", line 47, in read
    raise RuntimeError(self.packet_handler.getTxRxResult(comm_result))
RuntimeError: [TxRxResult] There is no status packet!
```

Which is the same error as I was originally encountering :(

@roncapat, perhaps you have an idea as to what's going on? I also had a question
concerning the implementation of bulk reading in `group_bulk_read.py`. The current implementation
allows for querying only one block of memory per motor. Is this a limitation in the Dynamixel SDK,
or just with the current implementation? I thought bulk reads would be useful to query several
different attributes from several different motors at once, and my reading of the documentation led
me to believe that was possible. The current code, however, does not allow this (explicitly at least:
it does look like I could extract sections of data if I wanted to). I just wanted to understand
what's going on.

Thanks for all your work on the project by the way!

roncapat, 2018-05-17:
hi @pipsqueaker :)

Let's see... Bulk read is documented here for protocol 1.0: [bulk_read](http://emanual.robotis.com/docs/en/dxl/protocol1/#bulk-read)
The documentation says that there's a limit of 1 block of memory per motor, and there are some deteils about how it works under the hood, dynamixel's side: 

> If an identical ID is designated multiple times, only the first designated parameter will be processed.

> When Bulk Read instruction is received, Dynamixel with ID 2 monitors the status packet being sent from ID 1 of the data bus (the preceeding device ID), and when device ID 1’s status packet transmission is completed, ID 2 sends its own status packet.

The python porting mimics exactly the behaviour of the C++ SDK, or at least it should, so group_bulk_read.py works as designed IMHO.

pipsqueaker, 2018-05-17:
It seems that I have made a huge oversight, and have been testing this code against AX-series motors; I think we can take this as confirmations that the bulk read command does, in fact, not work.

Thanks everyone for your time, and sorry for wasting it like this :|

closed 2018-05-17 by pipsqueaker

roncapat, 2018-05-17:
Well, it appears I have solved a problem without trying to solving it then 😂 

@ROBOTIS-Leon do you confirm that AX series do not support bulk_read?

roncapat, 2018-05-17:
@pipsqueaker no problem at all, we took the chance to learn a bit more about the behaviour of bulk read, so no wasted time :)

ROBOTIS-Leon, 2018-05-18:
@pipsqueaker 

Hi, and it appears like @roncapat says. 
I put related description in [the example code](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/989637569f458174831cfdaae194457821df3bc4/python/tests/protocol1_0/bulk_read.py#L26), but seems like hard to notice. 

The examples of protocol 1.0 sets MX series as default, since bulkread supports from MX series, and to make user not confuse of the model.