ROBOTIS-GIT/DynamixelSDK #190: [3.6.0][c] Issue: printTxRxResult[12] and printRxPacketError[12] are missing in linux64 library

Nungu, 2018-05-19:
I am using DynamixelSDK-3.6.0
I am trying to build the linux64 library
I get the following output:

```
mathias@mona:~/tmp/DynamixelSDK-3.6.0/c/build/linux64$ make clean && make
rm -f ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o ./libdxl_x64_c.so
mkdir -p ./.objects/
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_read.c -o .objects/group_bulk_read.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_write.c -o .objects/group_bulk_write.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_read.c -o .objects/group_sync_read.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_write.c -o .objects/group_sync_write.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/packet_handler.c -o .objects/packet_handler.o
../../src/dynamixel_sdk/packet_handler.c: In function ‘printTxRxResult’:
../../src/dynamixel_sdk/packet_handler.c:73:5: warning: implicit declaration of function ‘printTxRxResult1’ [-Wimplicit-function-declaration]
     printTxRxResult1(result);
     ^
../../src/dynamixel_sdk/packet_handler.c:77:5: warning: implicit declaration of function ‘printTxRxResult2’ [-Wimplicit-function-declaration]
     printTxRxResult2(result);
     ^
../../src/dynamixel_sdk/packet_handler.c: In function ‘printRxPacketError’:
../../src/dynamixel_sdk/packet_handler.c:95:5: warning: implicit declaration of function ‘printRxPacketError1’ [-Wimplicit-function-declaration]
     printRxPacketError1(error);
     ^
../../src/dynamixel_sdk/packet_handler.c:99:5: warning: implicit declaration of function ‘printRxPacketError2’ [-Wimplicit-function-declaration]
     printRxPacketError2(error);
     ^
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler.c -o .objects/port_handler.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol1_packet_handler.c -o .objects/protocol1_packet_handler.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol2_packet_handler.c -o .objects/protocol2_packet_handler.o
gcc -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler_linux.c -o .objects/port_handler_linux.o
g++ -shared -fPIC -m64 -o ./libdxl_x64_c.so ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o -lrt
```

The library installs without issues.

The C example `DynamixelSDK-3.6.0/c/example/protocol1.0/ping/linux64` doesn't build either because of the missing functions.

Grepping through the source tree I can't find any declaration of printTxRxResult1, printRxPacketError1, printTxRxResult2 and printRxPacketError2.

Nungu, 2018-05-19:
If I add these functions to ping.c it works.

AchmadFathoni, 2018-05-20:
It is deprecated function and already removed (poorly).

robotpilot, 2018-05-20:
Thanks, @AchmadFathoni 

Hi @Nungu,
Yes. The function was a deprecated function and has been completely removed in this 3.6.0 release.

robotpilot, 2018-05-27:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-05-27 by robotpilot

Nungu, 2018-05-28:
The library references the depricated functions, not my code. The examples that don‘t run are part of the library. You schould remove the references to these functions.

wyd0817, 2021-07-24:
Why is this function not recommended?

ROBOTIS-Will, 2021-07-26:
It is deprecated and removed in the API. 
You may use getTxRxResult() instead.