ROBOTIS-GIT/DynamixelSDK #191: [3.6.0][Python][Robotis OP2][MX28] Issue: Protocol 2 ping.py not working on updated mx28 motors?

pipsqueaker, 2018-05-24:
Hello,

I am attempting to communicate with my MX28 motors using protocol 2. I have updated them to firmware version 40, so this should be doable. However, when I run [the example code provided in the repository](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/ping.py) (with the baud rate changed to what I know to be correct), I get the following error

```
(dxl) robotis@robotis:~/DynamixelSDK/c/example/protocol2.0/ping/linux32$ python /tmp/thing2.py
Succeeded to open the port
Succeeded to change the baudrate
[TxRxResult] There is no status packet!
```

If I change `PROTOCOL_VERSION = 2.0` to `PROTOCOL_VERSION = 1.0` then the code runs perfectly. None of the other protocol 2 test code works either

Does anyone have any insight on this? I am happy to provide more information as needed

robotpilot, 2018-05-28:
Hi @pipsqueaker

I'm sorry for the late reply.

I will proceed with the DynamixelSDK test in the environment you mentioned. I will contact you as soon as a problem is found.

robotpilot, 2018-05-28:
I also tested it in a development environment like yours. However, I could not find a problem. Make sure the baudrate is not a problem. The default ping.py is set to 57600.

- MX28 / protocol 2

```bash
DynamixelSDK/python/tests/protocol2_0 $ python ping.py 
Succeeded to open the port
Succeeded to change the baudrate
[ID:001] ping Succeeded. Dynamixel model number : 30
```
```bash
DynamixelSDK/python/tests/protocol2_0 $ python read_write.py 
Succeeded to open the port
Succeeded to change the baudrate
Dynamixel has been successfully connected
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:010  PresPos:426
[ID:001] GoalPos:010  PresPos:416
[ID:001] GoalPos:010  PresPos:399
[ID:001] GoalPos:010  PresPos:374
[ID:001] GoalPos:010  PresPos:349
[ID:001] GoalPos:010  PresPos:322
[ID:001] GoalPos:010  PresPos:295
```

closed 2018-06-03 by robotpilot