ROBOTIS-GIT/DynamixelSDK #192: pan_tilt_port: No motors found + Dynamixel XM-430

pedrohgouvea, 2018-05-25:
Hello Everyone, I'm following the tutorial dynamixel_controllers/Tutorials/ConnectingToDynamixelBus and I cannot found my motor XM430

When I ran the roslaunch command I receive this:

> [FATAL] [1526461074.024123]: pan_tilt_port: No motors found.
================================================================================REQUIRED process [dynamixel_manager-2] has died!
process has died [pid 2580, exit code 1, cmd /opt/ros/kinetic/lib/dynamixel_controllers/controller_manager.py __name:=dynamixel_manager __log:=/home/odroid/.ros/log/35bcd1e4-58e7-11e8-abea-001e063102cf/dynamixel_manager-2.log].
log file: /home/odroid/.ros/log/35bcd1e4-58e7-11e8-abea-001e063102cf/dynamixel_manager-2*.log
Initiating shutdown!
================================================================================

My parameters are:

> PARAMETERS
>  * /dynamixel_manager/namespace: dxl_manager
>  * /dynamixel_manager/serial_ports/pan_tilt_port/baud_rate: 9600
>  * /dynamixel_manager/serial_ports/pan_tilt_port/max_motor_id: 5
>  * /dynamixel_manager/serial_ports/pan_tilt_port/min_motor_id: 1
>  * /dynamixel_manager/serial_ports/pan_tilt_port/port_name: /dev/ttyUSB0
>  * /dynamixel_manager/serial_ports/pan_tilt_port/update_rate: 20
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.1

Thanks a lot

robotpilot, 2018-05-27:
The [dynamixel_controllers](http://wiki.ros.org/dynamixel_controllers) you mentioned are user-created packages, not official ones provided by ROBOTIS. 

If you need the pan_tilt example, I recommend using the following dynamixel_workbench using the official DynamixelSDK.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#position-controller

pedrohgouvea, 2018-05-28:
Hi @robotpilot 
I'm trying to control a robot arm with 3 Dynamixel servos in the arm and 1 for the pincher.
I'm trying to follow this tutorial:
[](http://wiki.ros.org/dynamixel_controllers/Tutorials/ConnectingToDynamixelBus)

And if possible finish with this : [](http://wiki.ros.org/dynamixel_controllers/Tutorials/Controlling%20the%20PhantomX%20Pincher%20Robot%20Arm)

The dynamixel_workbench tutorials is not enough for me to understand how to control the robot arm.

Do you know any other tutorials, or can you give me advice about how to control the arm.

Thanks a lot.

robotpilot, 2018-05-28:
Hi @pedrohgouvea

If you need to use PhantomX, you can get help by posting related questions on the forum. The following address is a forum for companies developing and selling PhantomX.
http://forums.trossenrobotics.com/content.php

Also, the tutorial you mentioned seems to be related to GitHub below.
https://github.com/arebgun/dynamixel_motor

The ROS packages and tutorials you mentioned are not the software and content we manage. Please get answers from the developers and forums.

I will close this issue. If you have any issues with DynamixelSDK, please upload the issue again. 

Thanks.

closed 2018-05-28 by robotpilot