ROBOTIS-GIT/DynamixelSDK #197: [3.6.0][Python][USB2DYNAMIXEL][MX-64AR] Issue: read_write.py doesn't work due to "out of range" error

clarelywork, 2018-06-13:
- I'm using `DynamixelSDK ver. 3.6.0`

- I'm using `Python` Language

- I'm using `USB2Dynamixel2` serial converter

- I'm using `mx-64 AR`

- and I'm having an `issue` like what

              Sample script "read_write.py" won't run due to "Out of range" error

Hello!

I am running Dynamixel 3.6.0 in Python on my Linux laptop. The MX-64 motor is protocol 1, baud rate not changed (57142 according to Roboplus), motor is completely new from the box.

I am having trouble running the read_write.py under python/tests/protocol1_0:

Issue #1 (solved)

The sdk import (line 50) was not defined, which will run into an import error, by adding this line in line 49:

               [os.sys.path.append('../../src') ]

This issue can be solved.

Issue #2 (MAIN)

After the path is defined, the script will run but the command prompt will show this: 

[image]

(the present position value is sometimes at 000 as well)
To clarify, nothing else on the script is changed except the line stated in Issue #1. 
I have ran this multiple times and tried with different motors. I had another person run it (independent of my instructions), and we both ran into this issue.

Please let me know how I can fix this so that the sample script can run smoothly with library 3.6.0.

Thank you!

routiful, 2018-06-14:
Hi :)

Before you run example, you are supposed to set control table item in line [from 52 to 54](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol1_0/read_write.py#L52)

For example, `ADDR_MX_TORQUE_ENABLE` of MX-64 is `24` ([Ref](http://emanual.robotis.com/docs/en/dxl/mx/mx-64/#initial-value))

After replace others, you can run your Dynamixels.

Thanks

clarelywork, 2018-06-14:
Hello!

That worked! Thank you so much!
Regarding on the issue #1 noted on in the post, is that line required? without that line I keep running into import error.

Another not-so-related question: 
I am having trouble changing the baud rate, I am running read_write in the same setup, I have changed the mx-64 motor to 200,000 (9), and changed the baudrate value in line 62. However I am running into this issue as shown in attached image: 

[image]

Please let me know how I can set the new baud rate successfully. 
Thank you!

routiful, 2018-06-15:
Hi :)

No, before you run python code, you are supposed to install setup.py
Below link shown how to install it

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/#building-the-library

Moreover, `Dynamixel SDK` doesn't support 200,000 baudrate. I highly recommend 57600 or 1000000(1M)

Thanks

clarelywork, 2018-06-15:
Hello!

That clears things up, thank you for your help!

routiful, 2018-06-17:
Great. If you have any problem or inquiry about SDK, please feel free to open any issue.

Thanks :)

closed 2018-06-17 by routiful

ROBOTIS-Will, 2022-07-12:
@alinena 

You may use the min / max position value of your DYNAMIXEL.
The THRESHOLD value in the code is simply to determine whether the given position is reached or not (to send the next position value).

If you are getting the out of range error, please check if the data you are transmitting is within the valid range defined in the control table.
It is difficult for us to determine which part of the code causes the error without the source code and the error, and the DYNAMIXEL model you use.

Thanks!

alinena, 2022-07-12:
I'm using MX-28AT dynamixel and I'm trying the protocol1_0 of the librairy DynamixelSDK_37-31
And if I choose the minimum postion to 0 and 4095 for the max I don't have the "out of range error" anymore but the motor isn't running. 
[image]

alinena, 2022-07-12:
Now the motor is running but I have some issues to read the postion. 
The position printed in the Terminal is wrong (very big) and doesn't really make sense, and the value of the position become really big when I tried whith a bigger velocity, when the velocity is small the position printed is sometimes right. 
[image]

Thank you for your help!

ROBOTIS-Will, 2022-07-14:
Hi @alinena 

The MX-28AT with Protocol 1.0 firmware uses different control table then the Protocol 20.
The current read_write example is written for the X series, you should modify the Control table address definitions as below.

Thanks!

```python
# Control table address
ADDR_MX_TORQUE_ENABLE      = 24               # Control table address is different in Dynamixel model
ADDR_MX_GOAL_POSITION      = 30
ADDR_MX_PRESENT_POSITION   = 36
```

alinena, 2022-07-18:
Thank you, I find the problem to print the position. The position was wrong and big because, I was considering that the position length was 4 and in reallity for the MX-28AT it is 2 bytes.