ROBOTIS-GIT/DynamixelSDK #198: [3.6.1][C++][USB2Dynamixel2][MX106R] Issue : My motor doesn't work

lijing94123, 2018-06-14:
Hi, 

I was trying to run the basic "read_write" example by c++. After following the instructions setp by step and it actually was successfully compiled, however, when I finally started to run without debugging, there appears what you could see from the figure, just no dynamixel motor seem to be connected.

By the way, I've checked the motor by dynamixel wizard, it worked well.
[image]

lijing94123, 2018-06-15:
Hi, thank you for paying attention to my question!

And I have a few things to add. The package I used should be DynamixelSDK 3.6.0. I thought I have installed the newest one which is shown as 3.6.1. but then I found what I have downloaded is the same as 3.6.0.

And I have tried with othe packages. The first package I used before was DynamcelSDK 3.5.2, which I used late last year and earlier this year. Then I used it agaein. And it turns out that DynamixelSDK 3.5.2 worked well. My motors work now! I am wondering whether there could be any conflicts between different packages....

Thanks again and I still want to know what's wrong with my motor with DynamixelSDK 3.6.0.

robotpilot, 2018-06-15:
Hi @lijing94123 

Version-specific changes can be found in the following documents: 
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/ReleaseNote.txt

The major update of 3.6.0 version was **Python** support.
The 3.6.1 version is a minor update, and the C ++ example has been revised.

closed 2018-06-24 by robotpilot