ROBOTIS-GIT/DynamixelSDK #199: [3.6.0][Python][USB2DYNAMIXEL][MX-64] Issue: Unclear type error in C folder

clarelywork, 2018-06-15:
- I'm using `DynamixelSDK ver. 3.6.0`
- I'm using `Python` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `MX-64AR`
- and I'm having an `issue` like what
   `conflicting input type causing the script to crash`

Hello!

I am writing a motor script for my robot, which uses 4 MX-64 motors.
while reading the present value, I keep running into this error: 

[image]

I have other scripts that I use this exact line, and was able to read value and execute smoothly, however, this script never passes this function.

`portHandler = `PortHandler(DEVICENAME)`

`dynamixel` = `PacketHandler(PROTOCOL_VERSION)`

`def readPos():`

     `    pos1 = dynamixel.read2ByteTxRx(portHandler, DXL1_ID, ADDR_MX_PRESENT_POSITION)`
     `    pos2 = dynamixel.read2ByteTxRx(portHandler, DXL2_ID, ADDR_MX_PRESENT_POSITION)`
     `    pos3 = dynamixel.read4ByteTxRx(portHandler, DXL3_ID, ADDR_MX_PRESENT_POSITION)`
     `    pos4 = dynamixel.read4ByteTxRx(portHandler, DXL4_ID, ADDR_MX_PRESENT_POSITION)`
     `    return [pos1, pos2, pos3, pos4]`

Please let me know how I can solve this issue. Thank you!

routiful, 2018-06-18:
Hi :)

`read2ByteTxRx` function returns tuple including data_read, result, and error. 
So you are supposed to get return data to proper type. Below link might be help you

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/src/dynamixel_sdk/protocol1_packet_handler.py#L377

Thanks

clarelywork, 2018-06-18:
Hello!

I tried reading the tuple like this: 

 `    pos1 = dynamixel.read2ByteTxRx(portHandler, DXL1_ID, ADDR_MX_PRESENT_POSITION)[0]`

But the error is still listed: 

[image]

I have another script where this line runs as expected, but this script would show this error and crash.
Please let me know how I can fix this.

Thank you!

clarelywork, 2018-06-18:
Sorry I accidentally clicked on the close issue button

routiful, 2018-06-20:
I assume that the `port_handler,py` is not working properly. 
In your code, `port_handler` was gotten DEVICE_NAME in  `dbottest_v360.py`.
So, you are supposed to open it in `movement_v360.py`

Try this and ask me further
Thanks

robotpilot, 2018-06-26:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2018-06-26 by robotpilot